raspberrypi/bookworm-feedback #84: Visual Studio ssh connection

pceru, 2023-10-17:
I can't connect from Ms Visual Studio 2022 to a raspi 3B+ bookworm OS 32 bit, for remote compile a Cmake program, using ssh private key authentication and empty passphrase (password auth works), all my other bullseye raspi can connect using the same private key. Ssh from windows with the same key works fine

6by9, 2023-10-17:
Which algorithm is your key generated with / how old are they?

The ssh-rsa algorithm has been dropped from OpenSSH as it is judged to be insecure - https://www.openssh.com/txt/release-8.2

As a test, you can add 
```
PubkeyAcceptedAlgorithms=+ssh-rsa
```
to the end of /etc/ssh/sshd_config, reboot, and see if you can connect. If so, then the best course of action is to create new keys which should use the updated (and secure) algorithms.

Otherwise you'll need to enable logging of the SSH initialisation to see why the key is being rejected. If it were a normal ssh session, then adding -v enables logging. I can't say how you'd do that from within Visual Studio.

pceru, 2023-10-17:
Thank you very much, the PubkeyAcceptedAlgorithms=+ssh-rsa makes the trick. Yes it was a old rsa key but I don't understand why it works outside the MsVS program using every ssh program I found on Win 11.
Again thank you.

closed 2023-10-17 by pceru

6by9, 2023-10-17:
Please note my other comment then:
> If so, then the best course of action is to create new keys which should use the updated (and secure) algorithms.

As per the OpenSSL changelog, the SHA-1 hash was considered hackable in 2020 for moderate expenditure. 3 years on and that cost will have fallen even further.