raspberrypi/bookworm-feedback #86: Clipboard issues

qrp73, 2023-10-17:
Steps to reproduce:
1) Start Text Editor (mousepad)
2) type 12345
3) select text 234
4) Press Ctrl+Instert to copy text into clipboard
5) Close Text Editor
6) Start new Text Editor
7) Press Shift+Insert to paste previously copied text.

Expected result: text "234" is inserted

Actual result: nothing happens - BUG

8) Right click on Text Editor to get context menu
9) Select menu item Paste

Expected result: text "234" is inserted

Actual result: nothing happens - BUG

10) Right click on Text Editor to get context menu
11) Select menu item Paste Special->Paste from History

Expected result: text "234" should be proposed to insert

Actual result: "No clipboard data" popuip appears - BUG

spl237, 2023-10-18:
Clipboard handling in Wayland is limited, again due to the security model. The contents of the clipboard are only retained for as long as the application from which they are copied remains running. So in this case, because you are closing the text editor and then re-opening it, the previously copied text is lost when the application is closed.

This is just the way Wayland works and is not fixable.

closed 2023-10-18 by spl237

qrp73, 2023-10-18:
But there is also another issues. For example Shift+Insert may insert garbage text sometimes (catch it a lot of times in lxterminal) or just do nothing, but if I do paste through context menu with mouse it works ok. I still didn't catch a way to reproduce it, because this is floating bug (probably depends on some conditions), so this bug is not included in this issue, but I wanted to add it later...

qrp73, 2024-09-05:
After all, I found how to fix it. 
Just install clipman and configure wayfire to use it:

```sudo apt install clipman```

then ```nano ~/.config/wayfire.ini```
```
[autostart]
clipman-restore = clipman restore
clipman-store = wl-paste -t text --watch clipman store
```

Works great. Just wonder why it missing from clean install?