raspberrypi/bookworm-feedback #87: Bookworm lifting USB current limit doesn't seem to work?

Tiger0215, 2023-10-18:
When using the PD power supply, the USB port output current still does not reach 1.6A. Is there any instruction to test the USB output current or release the USB current limit?
I've currently used PD power, and lifted the USB current limit, but it doesn't work:
[image]

ghollingworth, 2023-10-18:
Can you confirm:

* I'm assuming this is Pi 5?
* This is the official Raspberry Pi PD power supply?
* What does "vcgencmd get_config usb_max_current_enable" give
* What does the first boot screen say about the power supply (remove any booting media if you want to hang around on this screen for longer)

[image]

Tiger0215, 2023-10-18:
`pi@raspberrypi:~ $ vcgencmd get_config usb_max_current_enable
usb_max_current_enable=1
`
But when I use a 1A electronic load on the USB interface, the USB interface power supply will continue to restart, that is, the measured USB output capability remains at 600mA.

Tiger0215, 2023-10-18:
* I'm assuming this is Pi 5? --Y
* This is the official Raspberry Pi PD power supply? --N
* What does "vcgencmd get_config usb_max_current_enable" give -- "usb_max_current_enable=1"
* What does the first boot screen say about the power supply (remove any booting media if you want to hang around on this screen for longer) -- ""USB-PD 5000 mA""

ghollingworth, 2023-10-18:
What do you get with:

pinctrl 49

Tiger0215, 2023-10-18:
`pi@raspberrypi:~ $ pinctrl 49

49: op dh pd | lo // EN_MAX_USB_CUR/GPIO49 = output
`

Tiger0215, 2023-10-18:
`49: op dh pd | lo // EN_MAX_USB_CUR/GPIO49 = output`

Tiger0215, 2023-10-18:
When I connect an electronic load greater than 1A, the USB interface keeps restarting
I'm not sure if there is a command to monitor the output status of the USB interface. If there is, I can try to read it.

timg236, 2023-10-18:
> `49: op dh pd | lo // EN_MAX_USB_CUR/GPIO49 = output`

That looks correct.

pin43 will tell you whether USB OC is asserted

ghollingworth, 2023-10-18:
@timg236 @pelwell 

I've just tried adding:

```usb_max_current_enable=0```

in config.txt, I've confirmed with vcgencmd that

```
gordon@gordon@~$ vcgencmd get_config usb_max_current_enable
usb_max_current_enable=0
gordon@gordon@~$ pinctrl 49
49: op dl pd | lo // EN_MAX_USB_CUR/GPIO49 = output
```
Without the config.txt parameter, I get:

```
49: op dh pd | lo // EN_MAX_USB_CUR.GPIO49 = output
```

Is there something wrong with this output?  In this case it is driving high (dh), but reading back 'lo'

Gordon

Tiger0215, 2023-10-18:
`49: op dh pd | hi // EN_MAX_USB_CUR/GPIO49 = output
`
Yes, I can see that the IO port is in a high level state.
But unfortunately, the actual USB output capability remains unchanged.

pelwell, 2023-10-18:
OK - I think the pinctrl output is red herring. The read value is wrong, but that's because the input enable has not been set on the pin. Running `sudo busybox devmem 0x1f000f8040 32 0x56` changes that, after which pinctrl shows that the pin level is high, as expected.

I can either display `--` for the level, or force the input enables to always be high - I think the former is safer and more informative.

pelwell, 2023-10-18:
FYI, `pinctrl` has been updated with the change described above: https://github.com/raspberrypi/utils/commit/4f4c1744591a909bbc389428e64ba999f6d2f2c8
```
pi@raspberrypi:~$ pinctrl 49
49: op dl pd | -- // EN_MAX_USB_CUR/GPIO49 = output
```

timg236, 2023-10-18:
@Tiger0215 Please could you post a photo of the underside of the board? High res around the RP1 area would be good to rule out some things

closed 2023-11-16 by Tiger0215