raspberrypi/bookworm-feedback #94: Netboot - /etc/resolv.conf isn't updated with the DHCP provided DNS servers

trejan, 2023-10-18:
/etc/resolv.conf isn't updated with the DHCP provided DNS servers when netbooting as NetworkManager considers eth0 to be externally managed. Kernel IP-Config shows the DNS servers but it isn't passed through to the system and the resolver will use the existing contents of resolv.conf. For a stock image, this will be 8.8.8.8 as it is part of the shipped image.

Bullseye uses openresolv which does update /etc/resolv.conf when netbooting. Bookworm doesn't install openresolv and adding it doesn't help as presumably its hook isn't being called by NM.

/etc/resolv.conf does get updated by NetworkManager when booting from local storage.

reraikes, 2023-10-18:
I can confirm tejan's findings.

gargoyle, 2024-07-05:
Also hit upon this today. 

Net booting with `ip=dhcp` in the `boot\cmdline.txt` file. However, nmcli reports that the connection is manually managed (full output at the end). However, the pi *IS* getting its statically assigned IP address from the DHCP server.

As a temporary work-around, is the best option to just edit `resolv.conf` manually (I assume it won't get overwritten if NM thinks it's manually configured)?

```
root@metrics:/etc/NetworkManager# nmcli c show eth0
connection.id:                          eth0
connection.uuid:                        9f9c5ad5-e872-45df-a914-005dd0343ce3
connection.stable-id:                   --
connection.type:                        802-3-ethernet
connection.interface-name:              eth0
connection.autoconnect:                 no
connection.autoconnect-priority:        0
connection.autoconnect-retries:         -1 (default)
connection.multi-connect:               0 (default)
connection.auth-retries:                -1
connection.timestamp:                   1720178285
connection.read-only:                   no
connection.permissions:                 --
connection.zone:                        --
connection.master:                      --
connection.slave-type:                  --
connection.autoconnect-slaves:          -1 (default)
connection.secondaries:                 --
connection.gateway-ping-timeout:        0
connection.metered:                     unknown
connection.lldp:                        default
connection.mdns:                        -1 (default)
connection.llmnr:                       -1 (default)
connection.dns-over-tls:                -1 (default)
connection.mptcp-flags:                 0x0 (default)
connection.wait-device-timeout:         -1
connection.wait-activation-delay:       -1
802-3-ethernet.port:                    --
802-3-ethernet.speed:                   0
802-3-ethernet.duplex:                  --
802-3-ethernet.auto-negotiate:          no
802-3-ethernet.mac-address:             DC:A6:32:1D:1E:16
802-3-ethernet.cloned-mac-address:      --
802-3-ethernet.generate-mac-address-mask:--
802-3-ethernet.mac-address-blacklist:   --
802-3-ethernet.mtu:                     auto
802-3-ethernet.s390-subchannels:        --
802-3-ethernet.s390-nettype:            --
802-3-ethernet.s390-options:            --
802-3-ethernet.wake-on-lan:             default
802-3-ethernet.wake-on-lan-password:    --
802-3-ethernet.accept-all-mac-addresses:-1 (default)
ipv4.method:                            manual
ipv4.dns:                               --
ipv4.dns-search:                        --
ipv4.dns-options:                       --
ipv4.dns-priority:                      0
ipv4.addresses:                         192.168.66.32/24
ipv4.gateway:                           192.168.66.1
ipv4.routes:                            --
ipv4.route-metric:                      -1
ipv4.route-table:                       0 (unspec)
ipv4.routing-rules:                     --
ipv4.replace-local-rule:                -1 (default)
ipv4.ignore-auto-routes:                no
ipv4.ignore-auto-dns:                   no
ipv4.dhcp-client-id:                    --
ipv4.dhcp-iaid:                         --
ipv4.dhcp-timeout:                      0 (default)
ipv4.dhcp-send-hostname:                yes
ipv4.dhcp-hostname:                     --
ipv4.dhcp-fqdn:                         --
[... 133 lines cut ...]
ipv4.gateway:                           192.168.66.1
ipv4.routes:                            --
ipv4.route-metric:                      -1
ipv4.route-table:                       0 (unspec)
ipv4.routing-rules:                     --
ipv4.replace-local-rule:                -1 (default)
ipv4.ignore-auto-routes:                no
ipv4.ignore-auto-dns:                   no
ipv4.dhcp-client-id:                    --
ipv4.dhcp-iaid:                         --
ipv4.dhcp-timeout:                      0 (default)
ipv4.dhcp-send-hostname:                yes
ipv4.dhcp-hostname:                     --
ipv4.dhcp-fqdn:                         --
ipv4.dhcp-hostname-flags:               0x0 (none)
ipv4.never-default:                     no
ipv4.may-fail:                          yes
ipv4.required-timeout:                  -1 (default)
ipv4.dad-timeout:                       -1 (default)
ipv4.dhcp-vendor-class-identifier:      --
ipv4.link-local:                        0 (default)
ipv4.dhcp-reject-servers:               --
ipv4.auto-route-ext-gw:                 -1 (default)
ipv6.method:                            link-local
```

DavidCWGA, 2024-07-05:
My temporary workaround is just to hardcode the dns servers in /etc/resolv.conf in advance. Which isn't pretty, but it works as long as your DNS server has a stable address.

gargoyle, 2024-07-05:
Sorry if it messes up the thread, but I edited my previous comment. Seems changing the DNS and DHCP setup on my net enabled the PI's (and other systems on my net) to correctly self configure via DHCP.

XECDesign, 2024-07-08:
This might be more of a NetworkManager configuration question.

I am by no means an expert in the area, so I don't think I'll be of much help here. My understanding is that NM correctly detects that something else is managing the interface and to err on the side of caution, does not mess with it. There may be a way to change that behaviour but google isn't helping me here today.

If somebody finds a solution, I'm happy to add it. Or maybe, the solution belongs in the documentation rather than in Raspberry Pi OS itself. Either way, it's probably for somebody more familiar with NM and how the problem is solved in other distributions (if it is).