raspberrypi/bookworm-feedback #95: audio device dropping after hours of use

pidloop, 2023-10-18:
Bookworm with all latest update/upgrade on 1 MB Pi4.

Audio from USB device that worked perfectly under bullseye on same Pi4 now drops spontaneously after a few to tens of hours of proper operation. The app dies but restarting it does not help, the device must be unplugged/replugged before it will work again. This is the only USB device and the only audio device of any kind on the system.

FWIW, the only audio related reports via journalctl that I see are the following:

```
Oct 18 13:41:44 raspberrypi2 pipewire[2636]: spa.alsa: front:3: snd_pcm_drop No such device
Oct 18 13:41:44 raspberrypi2 pipewire[2636]: spa.alsa: front:3: close failed: No such device

```

lurch, 2024-07-23:
Are you still having this problem with the latest Bookworm image?

pidloop, 2024-07-23:
Thanks for asking. No problems since installing the July 4th 2024 release of RPI OS.

closed 2024-07-23 by lurch