raspberrypi/bookworm-feedback #96: systemd graphical-session.target not working (inactive dead)

gbrayut, 2023-10-18:
[graphical-session.target](https://www.freedesktop.org/software/systemd/man/latest/systemd.special.html#Special%20Passive%20User%20Units) is a target designed to be active whenever any graphical session is running so you can use it to install user session dependent graphical services. But it does not appear to currently work on Rasberry Pi OS Bookworm:

```
$ systemctl --user list-units --type target --all
  UNIT                          LOAD      ACTIVE   SUB    DESCRIPTION                   
  basic.target                  loaded    active   active Basic System
● blockdev@dev-mmcblk0p1.target not-found inactive dead   blockdev@dev-mmcblk0p1.target
● blockdev@dev-sda1.target      not-found inactive dead   blockdev@dev-sda1.target
● blockdev@dev-sda2.target      not-found inactive dead   blockdev@dev-sda2.target
  bluetooth.target              loaded    inactive dead   Bluetooth
  default.target                loaded    active   active Main User Target
  graphical-session.target      loaded    inactive dead   Current graphical user session
  paths.target                  loaded    active   active Paths
  shutdown.target               loaded    inactive dead   Shutdown
  sockets.target                loaded    active   active Sockets
  sound.target                  loaded    inactive dead   Sound Card
  timers.target                 loaded    active   active Timers

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
12 loaded units listed.
To show all installed unit files use 'systemctl list-unit-files'.
```

I believe the desktop session needs to use "BindsTo=graphical-session.target" to activate the graphical-sesison.target

spl237, 2023-10-19:
Under X, I see exactly the same:

` UNIT                     LOAD      ACTIVE   SUB    DESCRIPTION
  basic.target             loaded    active   active Basic System
● blockdev@dev-sda1.target not-found inactive dead   blockdev@dev-sda1.target
  bluetooth.target         loaded    inactive dead   Bluetooth
  default.target           loaded    active   active Main User Target
  graphical-session.target loaded    inactive dead   Current graphical user session
  paths.target             loaded    active   active Paths
  shutdown.target          loaded    inactive dead   Shutdown
  sockets.target           loaded    active   active Sockets
  sound.target             loaded    inactive dead   Sound Card
  timers.target            loaded    active   active Timers

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
10 loaded units listed.
To show all installed unit files use 'systemctl list-unit-files'.
`

A bullseye image doesn't even list that target in response to that command, but if it is directly queried with systemctl status, it is also reported as inactive/dead. So it looks to me as if this hasn't changed and has always been like this.

What are the actual symptoms you are seeing which this is causing other than the output of systemctl? What functional effect is it having?

XECDesign, 2023-10-19:
I don't fully understand all of this, but it looks like what OP is saying is true in the case when the desktop environment is started as a systemd user session.

https://people.debian.org/~mpitt/systemd.conf-2016-graphical-session.pdf

I'm guessing we're not actually doing any of that and just rely on lightdm to handle everything.

gbrayut, 2023-10-19:
I use `systemctl set-default multi-user.target` for systems in my homelab, and then use the isolate command to switch between multi-user and graphical targets when needed. The special targets like graphical-session and sound are often used in the Install section (`WantedBy=graphical-session.target`) of systemd user session service units (in `~/.config/systemd/user` or `/usr/lib/systemd/user/`). Gnome and Cinnamon have a bunch of custom [convenience targets](https://manpages.debian.org/experimental/cinnamon/gnome-session-cinnamon.1.en.html#systemd), and you can also use [conditions](https://www.freedesktop.org/software/systemd/man/latest/systemd.unit.html#Conditions%20and%20Asserts) like `ConditionEnvironment=WAYLAND_DISPLAY` to make configuring session services across desktop environments much easier.

I'm not sure which component is suppose to manage the targets, but having a `/usr/lib/systemd/user/wayfire-session.target` that BindsTo and After=graphical-session.target would probably be the first step. Then starting that target in either `/usr/bin/wayfire-pi` or as an [autostart entry in wayfire.ini](https://github.com/WayfireWM/wayfire-wiki/blob/master/Basic-plugins.md#autostart) should work. Not sure if there is an easy way to automatically stop the target without using systemd services for the desktop environment, but it can always be manually stopped if needed.

Looks like [XDG Autostart](https://wiki.archlinux.org/title/XDG_Autostart) entries in `~/.config/autostart` or `/etc/xdg/autostart` with `OnlyShowIn=wayfire` are another possible workaround (tested and those do currently work), but that use the `[Desktop Entry]` style configuration instead of normal systemd service units.

Sarrus, 2024-05-31:
I'm having this issue as well, I tried adding `systemctl start graphical-session.target` as an autostart in `wayfire.ini` but systemd won't let you start `graphical-session.target` with `systemctl` because the unit file contains `RefuseManualStart=yes`.

I've worked round it for now by having my custom unit check for a graphical session periodically. Moving forward it would be nice if systemd can recognise that a graphical session has started.

yoiang, 2025-05-11:
> I've worked round it for now by having my custom unit check for a graphical session periodically. Moving forward it would be nice if systemd can recognise that a graphical session has started.

sorry for the late reply, can you share an example of how you accomplished that?

Sarrus, 2025-05-12:
Honestly, it's nothing particularly fancy. The code I'm running just crashes out if it can't find a graphical session and then systemd restarts it. The unit file contains:

```ini
[Service]
Restart=always
RestartSec=5
```

I'm not a fan of having programs restart endlessly on failure so would prefer to be able to do this instead:

```ini
[Unit]
After=graphical-session.target
```

This would let me ensure that a graphical session is running before I start any graphical units.