raspberrypi/bookworm-feedback #101: XMoveWindow fails on raspios/Wayland but works ok on ubuntu/Wayland

qrp73, 2023-10-19:
###Steps To Reproduce:
1) Login to Wayland desktop
2) Create test-x11.c:
```
#include <X11/Xlib.h>
#include <X11/Xutil.h>
#include <stdio.h>

// gcc -o test-x11 test-x11.c -lX11

int main() {
    Display *display = XOpenDisplay(NULL);
    if (!display) {
        fprintf(stderr, "XOpenDisplay failed\n");
        return 1;
    }
    int screen = DefaultScreen(display);
    Window root = RootWindow(display, screen);
    Window window = XCreateSimpleWindow(display, root, 0, 0, 320, 240, 1, BlackPixel(display, screen), WhitePixel(display, screen));
    XMapWindow(display, window);
    XStoreName(display, window, "XMoveWindow test");

    XSelectInput(display, window, ExposureMask | KeyPressMask);
    XMapRaised(display, window);    
    
    XEvent event;
    while (1) {
        XNextEvent(display, &event);
        if (event.type == KeyPress) {
            KeySym key = XLookupKeysym(&event.xkey, 0);
            switch (key) {
                case '1':
                    printf("XMoveWindow(100,100)...\n");
                    XMoveWindow(display, window, 100, 100);
                    break;
                case '2':
                    printf("XMoveWindow(200,200)...\n");
                    XMoveWindow(display, window, 200, 200);
                    break;
            }
        }
    }
    XCloseDisplay(display);
    return 0;
}
```
2) compile it with ```gcc -o test-x11 test-x11.c -lX11```
3) Run it with ```./test-x11```
4) Select window 
5) Press keyboard keys 1

Expected result: window should move to coordinate 100,100

Actual result on ubuntu/Wayland: window is moved to coordinate 100,100 => OK

Actual result on raspios-bookworm/Wayland: nothing happens, XMoveWindow fails => BUG

6) Press keyboard keys 2

Expected result: window should move to coordinate 200,200

Actual result on ubuntu/Wayland: window is moved to coordinate 200,200 => OK

Actual result on raspios-bookworm/Wayland: nothing happens, XMoveWindow fails => BUG

###Testing environment:

ubuntu:
```
$ echo $XDG_SESSION_TYPE
wayland
```

raspios-bookworm:
```
$ echo $XDG_SESSION_TYPE
wayland
```

As you can see XMoveWindow function works ok on Wayland in ubuntu, but fails on Wayland in raspios-bookworm

### Conclusion

Broken XMoveWindow function in raspios-bookworm/Wayland leads to many issues in different apps. Especially it leads to out of display window issues when window needs to dynamically expand/collapse. It often happens for context menu (see bookworm/Wayland taskbar Network plugin or VLC media player for example) and for app window (for example in media players like VLC, in messengers and games).

On the other hand XMoveWindow function works ok with Wayland on other OS such as ubuntu.

qrp73, 2023-10-20:
just tested it on Linux Arch under Wayland, XMoveWindow also works as expected on Arch/Wayland.

The only system where XMoveWindow fails on Wayland is raspios-bookworm.

ghollingworth, 2023-10-20:
Does it work under upstream Wayfire?  Or Sway (both of which are based on wlroots), would identify which codebase is the problem.  

What is the Wayland implementation on Arch you're using?

spl237, 2023-10-20:
XMoveWindow *cannot* work on Wayland. It is an instruction to an X server. There is no equivalent in Wayland; there is no Wayland instruction to which it can be mapped. It is functionally a no-op in a Wayland environment.

If you are seeing XMoveWindow work on a Wayland environment, then it has to be being processed by an X server, which in this case is presumably an Xwayland instance running in the Wayland environment.

As far as I am aware, Xwayland is a standard component across all Wayland instances. In order for it to respond to an XMoveWindow instruction it would need to map it onto a layer-shell instruction in Wayland. This does not appear to happen in the version of Xwayland we are using, which is the standard Debian bookworm release.

So if this is indeed working on Ubuntu, then they have customised their Xwayland instance to remap XMoveWindow into a layer shell command, which would be a very non-trivial thing to do. Grepping the Xwayland source code for the version we are using, I can find no references to XMoveWindow, or indeed to layer shell.

pelwell, 2023-10-20:
In Bookworm I get:
```
pi@raspberrypi:~$ Xwayland -version
The X.Org Foundation Xwayland Version 22.1.9 (12201009)
X Protocol Version 11, Revision 0
```

popcornmix, 2023-10-20:
Ubuntu it's:
```
$ Xwayland -version
The X.Org Foundation Xwayland Version 22.1.8 (12201008)
X Protocol Version 11, Revision 0
```
so marginally older (but there may be different patches).

spl237, 2023-10-20:
The other possibility, on thinking about this further, is that Ubuntu have implemented some mechanism whereby an XMoveWindow instruction received by Xwayland is then passed back to the compositor, which then reacts by relocating the window. But this would, as far as I know, be an extension to the default behaviour requred by a Wayland compositor, and not standard functionality.

I don't know whether that potential mechanism should be in wlroots or in wayfire, but my suspicion is that its absence is due to an optional feature not having been implemented rather than this being a bug.

The OP might like to raise the issue at https://github.com/WayfireWM/wayfire/issues with the Wayfire developers to see what their position is.

qrp73, 2023-10-20:
Here is Xwayland version from linux arch where XMoveWindow works ok:
```
$ Xwayland -version
The X.Org Foundation Xwayland Version 23.2.1 (12302001)
X Protocol Version 11, Revision 0
```

Currently cannot check ubuntu, because I replaced it with latest raspios-bookworm and trying to compile wayfire...

qrp73, 2023-11-15:
not relevant anymore

closed 2023-11-15 by qrp73