raspberrypi/bookworm-feedback #102: Shift+Insert paste wrong text from clipoard

qrp73, 2023-10-20:
### Steps To Reproduce
1) Run firefox, open some page, for example: https://forums.raspberrypi.com/viewtopic.php?t=357538
2) Select some text on the page
3) Open Geany editor
4) Type 12345, select text and do right click and select "Copy" to copy text into clipboard
5) Open lxterminal
6) Press Shift+Insert to past text from clipboard

Expected result: text "12345" inserted from clipboard

Actual result: text selected (and not copied!) on firefox window is inserted - BUG

7) Right click  and select "Paste" to paste text from clipboard

Expected result: text "12345" inserted from clipboard

Actual result: text "12345" inserted from clipboard  - OK

pelwell, 2023-10-20:
What makes you think this should work as standard?

qrp73, 2023-10-20:
Sorry, what standard? I'm trying to paste from clipboard, but it paste garbage which even never copied to clipboard... That's the issue...

pelwell, 2023-10-20:
Are you saying this is a Firefox bug?

pelwell, 2023-10-20:
The behaviour can be more simply demonstrated just with LXTerminal.
1. Type 'pasteinsert`.
2. Select 'paste' and press Ctrl-Shift-C.
3. Select 'insert' (don't press any keys).
4. Press Ctrl-Shift-V - 'paste' appears.
5. Press Shift-Insert - 'insert' appears.

Once again, what makes you think that Shift-Insert should paste everywhere? Being able to insert the most recently selected text effectively gives you a second clipboard - why have two shortcuts that always do the same thing?

pelwell, 2023-10-20:
Here's an explanation for the current behaviour: https://unix.stackexchange.com/questions/178070/why-does-shiftinsert-paste-from-clipboard-in-some-applications-and-primary-in-o

TL;DR - it's for compatibility with other Terminal apps.

ghollingworth, 2023-10-20:
Agree this is a choice that gnome have made and then inconsistently applied!

closed 2023-10-20 by ghollingworth

qrp73, 2023-10-20:
thanks for explanation. Is there a way to configure Shift+Insert to always paste from CLIPBOARD? Or at least to configure terminal...

pelwell, 2023-10-21:
As Google will tell you, you can configure the terminal app to make Shift-Insert the Paste shortcut, but I don't think you can do it without disabling Shift-Ctrl-V (the more logical shortcut).

qrp73, 2023-10-28:
Yes, I done it, just replaced Shift+Ctrl+V with Shift+Insert. 
Ctrl+Insert and Shift+Insert are much more common shortcuts for copy/paste from clipboard on other OS.