raspberrypi/bookworm-feedback #103: Problems with playing commercial DVDs

kip-the-elder, 2023-10-20:
Some commercial DVDs that play fine on exactly the same setup on bullseye, don't play properly on bookworm. Somemost work just fine.

When you open the DVDs in VLC, the main menu, where there is one, works fine. But... when I try to play the movie itself, it's pix-elated and jumpy almost like you might expect if someone had spread honey on the disc. :roll: :lol:

I don't think that this is a VLC problem, because, Xine and KODI do exactly the same thing. (I installed Xine, handbrake and KODI to test it) Exactly the same.  I tried to rip a disk on bookworm with handbrake to see if that also showed the issue.  The output was unwatchable and the same as if watched in Xine/Kodi/VLC directly. If I rip the same disk on the Bullseye Pi, it then plays without any problem on the bookworm Pi.

If I swap the bookworm SD card out and replace with the bulleseye; everything works as it should do.

Setup is as follows.

Raspberry pi 400 with official power supply.
OS installed on Samsung EVOplus 64GB.
Logitech Wireless mouse
DVD drive is : Bus 001 Device 006: ID 1c6b:a223 Philips & Lite-ON Digital Solutions Corporation DVD Writer Slimtype eUAU108
The DVD drive is connected via a "Sarent" 4 port powered hub.
OS kept up to date

libdvdcss is installed as per

sudo apt install libdvd-pkg
sudo dpkg-reconfigure libdvd-pkg

I also tried installing it from github in case the repo wasn't up to date.
I also tried downgrading to the same version that I'm using on the working bullseye setup. (1.4.2-1)

Neither had any effect.

Apart from the OS, both setups are identical. It's the same physical machine. I'm just swapping the SD cards.

Other things I've tried

Rebuilt the OS again from scratch. Just the basic desktop, then just VLC to start with. No difference
Tried switching from Wayland to X11. No difference.
Tried the USB "usb-storage.quirks" bodge in cmdline.txt in case it was an issue with the USB..as it does seem noticeably slower to initially access the drive; though I didn't see why it would be seeing as nothing about the hardware has changed. It made no difference as I expected.
Tried both setups on another 400s. Same thing...Works on Bullseye, not on Bookworm.
Compared the config.txt and cmdline.txt files to see if there was anything obvious. Nope. nothing..
Swapped the VLC config files from the Bullseye machine to the bookworm machine. No difference.

Some of those that don't work
Oceans 8
Red Road
Unthinkable
Mission impossible Rogue Nation

These work fine
Master and Commander
Mad Max Thunderdome
The Matrix
Men in Black 3

popcornmix, 2023-10-20:
The only DVD I have to hand is from your working list (The Matrix), so can't easily reproduce.

I guess a kernel update is a (slight) possibility. What does `uname -a` report on each system?
We could use rpi-update to force one system to match the other.

popcornmix, 2023-10-20:
Just to clearly rule out any media players. Can you test with dvdbackup (`sudo apt install dvdbackup`)?
```
pi@pi4:~/dvd $ dvdbackup -i /dev/sr0 -F -p
libdvdread: Attempting to retrieve all CSS keys
libdvdread: This can take a _long_ time, please be patient
libdvdread: Get key for /VIDEO_TS/VIDEO_TS.VOB at 0x00058a70
libdvdread: Error cracking CSS key for /VIDEO_TS/VIDEO_TS.VOB (0x00058a70)
libdvdread: Elapsed time 1
libdvdread: Get key for /VIDEO_TS/VTS_01_1.VOB at 0x00058b93
libdvdread: Elapsed time 0
libdvdread: Get key for /VIDEO_TS/VTS_02_0.VOB at 0x000c190b
libdvdread: Elapsed time 0
libdvdread: Get key for /VIDEO_TS/VTS_02_1.VOB at 0x000c86d8
libdvdread: Elapsed time 0
libdvdread: Get key for /VIDEO_TS/VTS_03_1.VOB at 0x0035a1a8
libdvdread: Elapsed time 0
libdvdread: Found 3 VTS's
libdvdread: Elapsed time 1
Copying menu: 100% done (55/55 MiB)
Copying Title, part 1/6: 100% done (1024/1024 MiB)
Copying Title, part 2/6: 100% done (1024/1024 MiB)
Copying Title, part 3/6: 100% done (285/285 MiB)
Copying Title, part 4/6: 100% done (1024/1024 MiB)
Copying Title, part 5/6: 100% done (1024/1024 MiB)
Copying Title, part 6/6: 100% done (880/880 MiB)
pi@pi4:~/dvd $ ls The\ Matrix/VIDEO_TS/
VTS_02_0.BUP  VTS_02_0.IFO  VTS_02_0.VOB  VTS_02_1.VOB  VTS_02_2.VOB  VTS_02_3.VOB  VTS_02_4.VOB  VTS_02_5.VOB  VTS_02_6.VOB
```
The VOB files contain the video. Have a look at the md5sum's on bullseye and bookworm:
```
$ md5sum *.VOB
b965c5a09d5c3d372e9a536c9783e35c  VTS_02_0.VOB
7d4e2d9205e8b7c3d2885d2432e63b25  VTS_02_1.VOB
4a433f8ad27ee78aa2252a4da64a7171  VTS_02_2.VOB
891e515e9573b4802a5fee1565b3d890  VTS_02_3.VOB
57f889452ae1ca95a90c71eba2976449  VTS_02_4.VOB
3529e1df2edef6a9b30b575054324dc8  VTS_02_5.VOB
f74c7767a58ff8081a67623753127fae  VTS_02_6.VOB
```
If they are different (I assume they are), then that explains why the video doesn't play correctly.

kip-the-elder, 2023-10-20:
on Bullseye

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

On 

Bookworm

Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

I note that bookworm seems to be running an older kernel...so I double checked... That's right.

kip-the-elder, 2023-10-20:
Yes, sure, I'll try dvdbackup. I probably won't be able to do it until later today or tomorrow though. I have already tried, VLC, Kodi, Xine and handbrake.

One thing that I noticed was that it only seems to be newer films that fail(though by no means all new films). Older ones seem to be OK...Though obviously my testing has been somewhat limited. 

I wondered if there had been some underlying change to they way the encryption is done which is causing a problem somewhere....

pelwell, 2023-10-20:
The kernel naming has changed with Bookworm to match Debian, but if you read further along you'll see that the version is 6.1.54, i.e. newer than 6.1.21.

popcornmix, 2023-10-20:
On bookworm, `sudo rpi-update stable~`, and change `auto_initramfs=1` to `auto_initramfs=0` in config.txt,
and reboot. Then check `uname -a`. Hopefully it matches bullseye.

Does it change anything?

kip-the-elder, 2023-10-21:
> Just to clearly rule out any media players. Can you test with dvdbackup (`sudo apt install dvdbackup`)?
>
> ```
> pi@pi4:~/dvd $ dvdbackup -i /dev/sr0 -F -p
> libdvdread: Attempting to retrieve all CSS [...]

md5sum from bookworm

1d99f0b1c07db99f91752bbdc05c293c  VTS_01_0.VOB
e6ea1d9b9b5509a0bcddf0024e59b175  VTS_01_1.VOB
3ad0bfe02e691aa4e1a691b618e3e6f8  VTS_01_2.VOB
adf8589c1ba2d57276328e5fb3ba8e4d  VTS_01_3.VOB
0f564b8d5a8c658c8faef8621097c687  VTS_01_4.VOB
867d182d51501f0247f702fdf20b2cce  VTS_01_5.VOB

md5sum from bullseye

1d99f0b1c07db99f91752bbdc05c293c  VTS_01_0.VOB
126d687abd03e94ec119f58cb2370e56  VTS_01_1.VOB
24ac221370c9eee2f49449287a5dcf7a  VTS_01_2.VOB
b5d9507be1f2e3a5cf6ca927772eaaca  VTS_01_3.VOB
fd45060454a319c526b306538954681e  VTS_01_4.VOB
845ad07770af83c819f310f80db1fc4f  VTS_01_5.VOB

kip-the-elder, 2023-10-21:
> On bookworm, `sudo rpi-update stable~`, and change `auto_initramfs=1` to `auto_initramfs=0` in config.txt, and reboot. Then check `uname -a`. Hopefully it matches bullseye.
> 
> Does it change anything?

$ Uname -a 

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

ghollingworth, 2023-10-21:
Might be worth trying with kernelv8.img it’s possible the ripping is broken with 16k pages

kip-the-elder, 2023-10-21:
> Might be worth trying with kernel8.img it’s possible the ripping is broken with 16k pages

I'm not sure that I understand you.....I'm pretty sure that I'm using that already. Isn't that the default on 64bit?

If I rename it, the system doesn't boot...so I think I am.

It's not just ripping that is failing. Just reading/playing a DVD doesn't work

lurch, 2023-10-21:
> It's not just ripping that is failing. Just reading/playing a DVD doesn't work

They're effectively the same thing:
 - ripping the DVD reads the data from the disc and writes it to a file on your SD card
 -  playing the DVD reads the data from the disc and decodes it to video and displays it on your screen

But for some (as yet unknown) reason it looks like Bookworm isn't able to "properly" read the data from your Ocean's 8 DVD, which is both why you're getting corrupted playback and md5sum mismatches (compared to Buster) when ripping. The fact that you're able to successfully play your backup done in Buster within Bookworm, indicates that Bookworm has no problem with the "decodes it to video and displays it on your screen" part.

If you try doing multiple rips of Ocean's 8 on Bookworm, do you get the same (incorrect) md5sums each time (which would point to a systematic error) or do you get different md5sums each time (which would point to it being a slightly random error) ?

6by9, 2023-10-21:
16k pages are only relevant for Pi5, but this is Pi400 so not relevant.

6by9, 2023-10-21:
Looking at [libdvdcss2](https://packages.debian.org/bookworm/libdvdcss2), which is provided by [libdvd-pkg](https://packages.debian.org/bullseye/libdvd-pkg), Bullseye (and Buster) were on [version 1.4.2](https://packages.debian.org/bullseye/libdvd-pkg), but Bookworm is on [1.4.3](https://packages.debian.org/bookworm/libdvd-pkg).

Upstream repo for it is at https://salsa.debian.org/multimedia-team/libdvd-pkg.
The changelog doesn't have anything too significant looking
```libdvd-pkg (1.4.3-1-1) unstable; urgency=medium

  * Team upload

  [ Debian Janitor ]
  * Drop custom source compression.
  * Bump debhelper from old 9 to 12.

  [ Marcos Fouces ]
  * Add templates.pot file

  [ Sebastian Ramacher ]
  * debian/po/es.po: Add Spanish translation (Closes: 987343)
  * New upstream release (Closes: #989985)
  * libdvdcss/: Modernize packaging
  * debian/control: Bump debhelper compat to 13
  * debian/rules: Fix order of tar arguments

  [ Hideki Yamane ]
  * check with https
  * use https for the download (Closes: #865347)

 -- Sebastian Ramacher <sramacher@debian.org>  Tue, 07 Sep 2021 00:48:16 +0200
```

Functionality appears to be using [wget from](https://salsa.debian.org/multimedia-team/libdvd-pkg/-/blob/master/libdvdcss/debian/rules) https://download.videolan.org/pub/libdvdcss/$(VER)/libdvdcss-$(VER).tar.bz2 to download the relevant binaries (those aren't in the Debian package). Chasing therefore needs to head into the VLC repos.

You could try installing 1.4.2 manually from https://download.videolan.org/pub/libdvdcss/1.4.2/

kip-the-elder, 2023-10-22:
> Looking at [libdvdcss2](https://packages.debian.org/bookworm/libdvdcss2), which is provided by [libdvd-pkg](https://packages.debian.org/bullseye/libdvd-pkg), Bullseye (and Buster) were on [version 1.4.2](https://packages.debian.org/bullseye/libdvd-pkg), but Bookworm is on [1.4.3](https://packages.debian.org/bookworm/libdvd-pkg).
> 
> Upstream repo for it is at https://salsa.debian.org/multimedia-team/libdvd-pkg. The changelog doesn't have anything too significant looking
> 
> ```
> 
>   * Team upload
> 
>   [ Debian Janitor ]
>   * Drop custom source compression.
>   * Bump debhelper from old 9 to 12.
> 
>   [ Marcos Fouces ]
>   * Add templates.pot file
> 
>   [ Sebastian Ramacher ]
>   * debian/po/es.po: Add Spanish translation (Closes: 987343)
>   * New upstream release (Closes: #989985)
>   * libdvdcss/: Modernize packaging
>   * debian/control: Bump debhelper compat to 13
>   * debian/rules: Fix order of tar arguments
> 
>   [ Hideki Yamane ]
>   * check with https
>   * use https for the download (Closes: #865347)
> 
>  -- Sebastian Ramacher <sramacher@debian.org>  Tue, 07 Sep 2021 00:48:16 +0200
> ```
> 
> Functionality appears to be using [wget from](https://salsa.debian.org/multimedia-team/libdvd-pkg/-/blob/master/libdvdcss/debian/rules) https://download.videolan.org/pub/libdvdcss/$(VER)/libdvdcss-$(VER).tar.bz2 to download the relevant binaries (those aren't in the Debian package). Chasing therefore needs to head into the VLC repos.
> 
> You could try installing 1.4.2 manually from https://download.videolan.org/pub/libdvdcss/1.4.2/

Indeed.

That was one of the first thing I tried. Sadly, it didn't resolve the issue.

kip-the-elder, 2023-10-22:
This morning, I tried with the 32 bit OS.

Exactly the same.

kip-the-elder, 2023-10-24:
There were some updates this morning....

sudo apt list --upgradable
Listing... Done
gldriver-test/stable,stable 0.15 all [upgradable from: 0.14]
libfm-data/stable,stable 1.3.2-1+rpt8 all [upgradable from: 1.3.2-1+rpt7]
libfm-extra4/stable 1.3.2-1+rpt8 arm64 [upgradable from: 1.3.2-1+rpt7]
libfm-gtk-data/stable,stable 1.3.2-1+rpt8 all [upgradable from: 1.3.2-1+rpt7]
libfm-gtk4/stable 1.3.2-1+rpt8 arm64 [upgradable from: 1.3.2-1+rpt7]
libfm-modules/stable 1.3.2-1+rpt8 arm64 [upgradable from: 1.3.2-1+rpt7]
libfm4/stable 1.3.2-1+rpt8 arm64 [upgradable from: 1.3.2-1+rpt7]
pcmanfm/stable 1.3.2-1+rpt17 arm64 [upgradable from: 1.3.2-1+rpt16]
pi-package-data/stable,stable 0.14 all [upgradable from: 0.13]
pi-package-session/stable 0.14 arm64 [upgradable from: 0.13]
pi-package/stable 0.14 arm64 [upgradable from: 0.13]
raspberrypi-ui-mods/stable 1.20231023 arm64 [upgradable from: 1.20231017]
rpi-eeprom/stable,stable 20.0-2 all [upgradable from: 19.1-1]
wf-panel-pi/stable 0.51 arm64 [upgradable from: 0.50]

Something in that lot seems to have resolve the problem. I've only tested it on one DVD; Oceans 8...But that is definitely working now...mostly..... The main feature will now play in VLC and can be ripped with Handbrake (I just ripped 2 chapters and all is fine)...however, now the main DVD menu doesn't play now.  You have to select the film from the  playback menu. That works fine. Personally. I think that's OK...Who need menus..But I thought I should mention it anyway.

Well done to whoever fixed it.

I'd love to know what the problem was.

popcornmix, 2023-10-24:
> I'd love to know what the problem was.

So would I. I don't think there was a deliberate fix for this.
It's possible it has been accidentally fixed with some other change.

Or perhaps it's not been fixed. Maybe the issue only occurs or doesn't occur
after some specific sequence of operations (e.g order of apps accessing DVD drive).

Let us know if the fix seems permanent, and if it also fixes other DVDs.

kip-the-elder, 2023-10-24:
> So would I. I don't think there was a deliberate fix for this. It's possible it has been accidentally fixed with some other change.
>
> Or perhaps it's not been fixed. Maybe the issue only occurs or [...]

Yes....I didn't see anything in that list of updates that jumped out at me in a sort of..."Ahh...that'll be it!"  kind of way.

I think that the first is more likely...It wouldn't be the first time that I've seen "accidental" fixes to software in my career...I've never been known to take the credit though...oh no. ;-) ;-)

I've done exactly the same sequence as before in order to test. I'm pretty sure that nothing has changed except the update. I'm only using that system for testing.   Cold start. fresh installation. Nothing else running.

I will try a few others, to confirm that it's not just a "fluke"

kip-the-elder, 2023-10-24:
Ok...I've just tried two more of those that I know didn't work before..

MI - Rogue Nation and Unthinkable.

The first works fine. Even the menu plays properly. Unthinkable; the menu is garbled, but the film plays fine; just like Ocean's 8.

So, it definitely looks as though something has been changed.

One other thing is that it "seems" to be much slower to initially load DVDs than before...That's anecdotal; I haven't actually timed it. Could well be entirely unrelated of course.

kip-the-elder, 2023-10-24:
"That may be a positive sign. Taking time may mean it's actually cracking the keys, rather than skipping that and outputting non-decrypted data."

yes, that was my thoughts too.

"I wonder if you had a bad version of dvdcss,"

Unlikely,, I would think as I tried several complete, fresh rebuilds and also manually installed the version from Bullseye. I also tested it on the 32 bit OS as well.

I will go and see if deleting  that data makes any difference.

Watch this space

popcornmix, 2023-10-24:
One other factor is DVD region. Are all your DVDs from the same region? You can `apt install regionset`
```
pi@pi4:~ $ sudo regionset /dev/sr0 
Current drive parameters for /dev/sr0:
  RPC Type: Phase II (Hardware)
  RPC Status: no region code set (bitmask=0xFF)
  Vendor may reset the RPC 4 times
  User is allowed change the region setting 5 times
Would you like to change the region setting for this drive? [y/n]: 
```
[This thread](https://ubuntuforums.org/showthread.php?t=1574740) talked a bit about `.dvdcss` and regions. Note:
> When you play a dvd for the first time a folder is created in .dvdcss that contains the decryption keys.
All players will access .dvdcss and if a folder is found matching the volume label of the dvd, *they will then use the keys inside whether good, bad or incomplete*
When there is a region mismatch between the disk and the drive, - libdvdcss will switch from the default extraction method (disc key) to a brute force method which is usually successful but not always.
The most likely titles on a dvd to produce wrong or incomplete keys are small sample titles like extras.
So going back to what I've mentioned, if the drive used for the 'first play' on a dvd fails to get the proper, complete set then all subsequent plays of that title will use that 'bad' set, no matter what the player or drive used.

I'm guessing "small sample titles like extras" more likely to be wrong also applies to short video loops used in menus.

kip-the-elder, 2023-10-24:
> One other factor is DVD region. Are all your DVDs from the same region? You can `apt install regionset`
>
> ```
> pi@pi4:~ $ sudo regionset /dev/sr0
> Current drive parameters for /dev/sr0:
>   RPC Type: [...]

Yes. They are all the same region.

I haven't checked the extras. But in fact the reverse happened...The Menu worked but the movies didn't...My guess on that..was that perhaps the menu wasn't encrypted, only the main film was...

popcornmix, 2023-10-24:
> Then tried to play oceans 8 ....DIdn't work. It recreated the directory and generated the codes....But doesn't seem to work...Back to how it was before the update.....Now I'm confused. LOL

you aren't doing anything like mounting the same home folder on bullseye and bookworm?
Which could have the effect of bullseye creating the correct keys in `.dvdcss` making bookworm subsequently work (until you delete them).

Can you try deleting `.dvdcss` directories on bullseye/bookworm. Confirm that when playing DVD bullseye works and bookworm doesn't. Check the keys stored in `.dvdcss` are different on bullseye/bookworm. Then try swapping `.dvdcss` directories over. Does that stop bullseye working and make bookworm work?

kip-the-elder, 2023-10-24:
> you aren't doing anything like mounting the same home folder on bullseye and bookworm? Which could have the effect of bullseye creating the correct keys in `.dvdcss` making bookworm subsequently [...]

No. Completely separate systems. I am swapping the SD cards out. One is bullseye...which is our "live " system. The other is bookworm for testing only. No connection whatsoever. 

Just renamed the .dvdcss directory on Bullyseye, then tried to play it. As expected, it recreated the files and played fine....and as expected it took sightly longer to load than when it already had the key..

I'll try moving the bullseye .dvdcss over to the bullseye stick and vice versa.  It'll be tomorrow now...as I've just been told that it's way past beer-o'clock and I must stop now ;-)

kip-the-elder, 2023-10-25:
Done a few more things and made a few more discoveries....

Firstly...it seems that after the update...Most of the DVDs that didn't work on bookworm now do...If I delete the .dvdcss directory so that the code gets recalculated, most work now.  Ocean's 8 doesn't. which is odd. I haven't got the time to test very one of the umpteen DVDs in my collection, but all that I have tested now seem to work...;-)

I tried deleting the ocean's 8, dvdcss directory on the bulleseye installation and it was recreated correctly and payed quite fine. If I used the directory created on bookworm on bullseye, it doesn't work either. Delete it again so that bullseye recreates the key files and all is well...So something is clearly different....and it definitely looks to be related to the decryption process   There was another load of updates this morning that included some gstreamer stuff..Didn't seem to make any difference though. 

If I go into the relevant dvdcss directory on each setup, I get the same 4 files...which presumably correspond to each of the tracks on the Oceans 8 DVD?

0000000130  000000099b  0000001042  0000245385

If I look at the contents of each

$ cat *

I get 

On bullseye

c0:00:00:19:c9
c0:00:00:19:c9
c0:00:00:19:c9
c0:00:00:19:ca

on bookworm

c0:00:00:19:c9
c0:00:00:19:c9
74:93:02:7b:06
c0:00:00:19:ca

If I do the same thing for MI5...which was was one that didn't work, but now does, I get

on bulleseye

00:00:00:00:00
c0:00:00:00:01
c0:00:00:00:02
c0:00:00:00:02
c0:00:00:00:03
c0:00:00:00:03
c0:00:00:00:04
c0:00:00:00:05
c0:00:00:00:05
c0:00:00:00:06

on bookworm

00:00:00:00:00
5e:55:d0:d1:75
c0:00:00:00:02
c0:00:00:00:02
c0:00:00:00:03
c0:00:00:00:03
c0:00:00:00:04
c0:00:00:00:05
c0:00:00:00:05
c0:00:00:00:06

I note that on both, there is one file that is different...and yet everything seems to work on MI5, quite fine..I played every track on that one, just to be sure.

Hope that helps.

kip-the-elder, 2024-07-24:
Yes.

closed 2024-07-24 by lurch