raspberrypi/bookworm-feedback #104: No fall back to link local/mDNS for IP addresses

thagrol, 2023-10-20:
Unlike Buster and earlier, network interfaces attached to networks without a DHCP server do not fall back to self assigned IPv4 addresses in the link local (169.254.0.0/16) subnet.

Impacts at least the `eth?` intefaces and the `usb?` interfaces provided by the ethernet gadget.

Sample ifconfig -a output from a Pi5 connected to a 4B via an ethernet cable both running bookworm:

```
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether d8:3a:dd:7c:f9:24  txqueuelen 1000  (Ethernet)
        RX packets 44  bytes 10364 (10.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 109  bytes 18376 (17.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 107

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 125  bytes 10547 (10.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 125  bytes 10547 (10.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.99.21  netmask 255.255.255.0  broadcast 192.168.99.255
        inet6 fe80::91ab:3ed8:b470:858c  prefixlen 64  scopeid 0x20<link>
        ether d8:3a:dd:7c:f9:25  txqueuelen 1000  (Ethernet)
        RX packets 6423  bytes 1511237 (1.4 MiB)
        RX errors 0  dropped 3  overruns 0  frame 0
        TX packets 2907  bytes 414253 (404.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Does not apprear to be specific to the Pi model. Same behaviour on a Pi 5, 4B, and zeroW.

thagrol, 2023-10-20:
Just to add, in all cases no changes from the default have been made to networking configuration. Whether booting to Desktop or command line the result is the same.

thagrol, 2023-10-21:
Ugly, hacky, workaround: https://forums.raspberrypi.com/viewtopic.php?p=2147672#p2147672

thagrol, 2023-10-24:
zero-conf. Not mDNS.

ghollingworth, 2023-10-25:
You should use a lower priority static network assignment

https://askubuntu.com/questions/1186636/how-to-configure-a-fallback-static-ip-address-when-there-is-no-dhcp-server-prese

closed 2023-10-25 by ghollingworth

thagrol, 2023-10-25:
How does one do that on a headless Pi where no other Linux machine is available to allow mounting and editing of files on the root partition?

It is not unreasonable to expect Bookworm to behave the same way as previous releases. Further, if the above is the official position going forward, this should be added to the official documentation.

XECDesign, 2023-10-25:
I don't have time to look at it properly, but based on the discussion [here](https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/966), it sounds like there are some runes which can be put into `NetworkManager.conf` that would make it work.

If anybody wants to give it a go, the relevant documentation seems to be [here](https://networkmanager.dev/docs/api/latest/nm-settings-nmcli.html).

thagrol, 2023-10-25:
Nope. I've tried setting ipv4.link-local both in /etc/NetworkMagager/NetworkManager.conf and on a single connection via nmcli

ghollingworth, 2023-10-25:
Worked fine for me...

nmcli connection modify Wired\ connection\ 1 ipv4.method link-local ipv4.link-local enabled

thagrol, 2023-10-25:
Does for me now. Must have been doing something wrong the first time.

Now to work out:
1. How to make that permanent
2. How to make that a fall back behaviour
3. How to apply that to a headless setup without the need to edit file sin the root partition

ghollingworth, 2023-10-25:
Just add it as a first run.sh same way as imager sets anything up…

thagrol, 2023-10-27:
I believe I have an alpha solution. See https://github.com/thagrol/nm/blob/main/nm-setup.sh and https://forums.raspberrypi.com/viewtopic.php?t=358527