raspberrypi/bookworm-feedback #105: again - browsers unusable on RPi model < 4B, reason discovered

sgalke, 2023-10-21:
just the same symptom as described in #89  again yesterday!
browsers freeze the machine after a few seconds...
reason:   wayland  selected in raspi-config dialog on models < 4B!

sgalke, 2023-10-21:
.. vnc connection seems to trigger also  browser freezing the system.

popcornmix, 2023-10-21:
> reason: wayland selected in raspi-config dialog on models < 4B!

Are you saying you have manually enabled wayland on a Pi3 (or lower)?
Or are you saying wayland was enabled automatically? (it shouldn't be).

We've not enabled wayland by default on those devices because the memory consumption is too high.
It wouldn't be surprising that browsers are not usable in a low memory situation.

sgalke, 2023-10-21:
manually enabled wayland in raspi-config

lurch, 2023-10-22:
Running Wayland isn't officially recommended (yet) on a Pi3 or below, so closing this issue.

closed 2023-10-22 by lurch