raspberrypi/bookworm-feedback #107: Pi5 16K PageSize Kernel Incompatible Software

theofficialgman, 2023-10-21:
This issue will be a public tracker for 16K pagesize incompatible software. I made a similar post in the bookworm beta forum thread but that is still not public.

Feel free to add your own incompatible software as a comment and I will add it to this post.

Much of this list is thanks to Asahi linux wiki https://github.com/AsahiLinux/docs/wiki/Broken-Software

Broken software:
- f2fs
- FEX
- hardened_malloc
- jemalloc
  - Geekbench 6
  - Minecraft Java (on versions and launchers without the lwjgl native jemalloc patch)
  - rust
- fd
- Wine ARM64
- Wine x86_64 through Box64
- Zig
- All Chromium based applications (pre-102)
- qt5-webengine
- qt6-webengine (QtPdf)
- libvirt/QEMU/KVM (fixed in versions newer than bookworm has)
- All Android applications (only 4k header compatible, https://reviews.llvm.org/D55029)
  - Minecraft Bedrock Launcher
- Most armv6/armv7 software

ghollingworth, 2023-10-21:
If you find this is a problem, switch to the 4k page size using:

kernel=kernelv8.img

in configuration.txt

closed 2023-10-21 by ghollingworth

popcornmix, 2023-10-21:
Scratch 3. https://github.com/raspberrypi/bookworm-feedback/issues/91

theofficialgman, 2023-10-26:
unable to systemd-nspawn 32bit image https://github.com/raspberrypi/bookworm-feedback/issues/120

retropieuser, 2023-11-08:
Flycast

lurch, 2023-11-10:
btrfs-convert #136

Botspot, 2023-11-14:
MuseScore. https://github.com/Botspot/pi-apps/issues/2480

Botspot, 2023-12-01:
Minecraft Pi

theofficialgman, 2023-12-02:
> > Minecraft Pi
> 
> Doesn't minecraftPi use Dispmanx for its display, or has that been updated? If it is still Dispmanx, then I am not sure it will work on Bookworm/Pi5 even with 4k pages.

I think what Botspot is referring to is the project based on Minecraft Pi edition called and Minecraft Pi Reborn. It still run the original binary but patches it (somehow, don't really know the technical details) to use more modern display stacks (I think glfw is used), add features and mods, and native arm64 support without armhf multiarch. I can confirm that Minecraft Pi Reborn works on Pi5 Bookworm with 4K pagesize.

JamesH65, 2023-12-02:
Thanks for the update, I did wonder what had happened to Minecraft Pi. Bizarre that an app like that has some dependency on page size.

pelwell, 2023-12-02:
Any app that tries to mmap something (hardware, for example, or perhaps RAM that is shared with the firmware) has to ensure that the base address that is mmaped is aligned to the OS page boundaries, and that alignment has now jumped from 4k to 16k.

theofficialgman, 2023-12-03:
> Bizarre that an app like that has some dependency on page size.

@pelwell Every single executable file has a dependency on page size (see asahi wiki page on the subject https://github.com/AsahiLinux/docs/wiki/Broken-Software#why-does-not-work-sometimes-mean-instantly-segfault). Older versions of the gold linker only produced 4K pagesize compatible binaries though this was corrected in 2015 https://sourceware.org/legacy-ml/binutils-cvs/2015-10/msg00050.html some old software is of course still affected by that gold linker issue even 2023. Some software manually overwrite the pagesize and that is the list of software you see above.

pelwell, 2023-12-03:
Understood (it wasn't my comment)

mnorrsken, 2023-12-18:
jemalloc - bitnami/redis docker image

Botspot, 2023-12-18:
Cura slicer AppImage https://github.com/smartavionics/Cura

dstumbo, 2023-12-19:
Polars (python module) because it uses jemalloc. ghollingworth's suggestion of using kernel8.img fixed it for me. I have read elsewhere that ubuntu uses a different page size and does not have the jemalloc issue, but I have not verified it.

Widge-5, 2024-01-16:
libretro flycast.
(the flycast already mentioned and fixed was flyinghead flycast known as lr-flycast-dev in retropie-setup. I'm referring specifically to lr-flycast which is still affected).

popcornmix, 2024-04-24:
Are you saying if you install it by downloading the linked deb file, it works with kernel8.img, but not with kernel_2712.img?
But if you build from source (presumably a newer version?) it works on both?

koverg70, 2024-04-24:
> Are you saying if you install it by downloading the linked deb file, it works with kernel8.img, but not with kernel_2712.img? But if you build from source (presumably a newer version?) it works on both?

Yes indeed!

popcornmix, 2024-04-25:
Sounds like it's best reported to noip that updating their deb files would solve this issue.

basisbit, 2024-05-27:
OvenMediaEngine, because it uses jemalloc (and the RaspberryPi 5 even with 8GB RAM doesn't have enough RAM to fully recomplie it and test if the jemallow configure option fixes all issues).

Botspot, 2024-05-27:
> OvenMediaEngine, because it uses jemalloc (and the RaspberryPi 5 even with 8GB RAM doesn't have enough RAM to fully recomplie it and test if the jemallow configure option fixes all issues).

Maybe try fewer threads, if possible. Also ZRAM should fix that for you. https://forums.raspberrypi.com/viewtopic.php?t=327238

lurch, 2024-09-30:
maxima #297

kmontocam, 2024-12-05:
Running fluent-bit containerized (have tested standalone) also

leolivier, 2024-12-20:
> jemalloc - bitnami/redis docker image

As well as any "internal" redis installation in images. I install redis in my own image like this:
```
RUN	curl -fsSL https://packages.redis.io/gpg | gpg --dearmor -o /usr/share/keyrings/redis-archive-keyring.gpg &&\
		echo "deb [signed-by=/usr/share/keyrings/redis-archive-keyring.gpg] https://packages.redis.io/deb $(lsb_release -cs) main" > /etc/apt/sources.list.d/redis.list &&\
		apt-get update &&\
		apt-get install -y redis	&&\
		...
```
and I get a `<jemalloc>: Unsupported system page size` when starting and redis crashes

csdivad, 2025-01-03:
What about mongo (tcmalloc)? Running its docker image results in
```
1 src/third_party/tcmalloc/dist/tcmalloc/system-alloc.cc:755] MmapAligned() failed - unable to allocate with tag (hint=0x4d80c0000000, size=1073741824, alignment=1073741824) - is something limiting address placement?
1 src/third_party/tcmalloc/dist/tcmalloc/arena.cc:48] CHECK in Alloc: FATAL ERROR: Out of memory trying to allocate internal tcmalloc data (bytes=131072, object-size=640); is something preventing mmap from succeeding (sandbox, VSS limitations)?
```

illwieckz, 2025-01-09:
Are affected too:

- Google Native Client (virtual machine):
  https://www.chromium.org/nativeclient/
- Daemon (game engine, it uses Native Client to virtualise the game code):
  https://github.com/DaemonEngine/Daemon
- Unvanquished (game, it uses the Dæmon game engine):
  https://unvanquished.net/

Here is the issue on Unvanquished side:

- https://github.com/Unvanquished/Unvanquished/issues/3286

The Native Client virtual machine technology is now deprecated by Google in favor of WebAssembly, unfortunately WebAssembly hasn't reached yet the level of features of Native Client and is then not usable yet for the needs of game engines like Dæmon. That is why the Dæmon game engine still relies on Native Client.

The Native Client loader on ARM is 32-bit only, Google never implemented it for 64-bit, so the 64-bit Dæmon game engine uses the 32-bit NaCl loader, the same way the 64-bit Google Chrome uses the 32-bit NaCl loader.

It is unlikely Google will distribute a new NaCl loader. We cannot exclude that a recompilation may not be enough to fix it, because the loader is known to do some very low-level tricks on memory to implement the isolation.

As a side note, I see that “All Chromium-based applications (pre-102)” are affected, it's possible that when Chromium is rebuilt to work for 16k PageSize kernels, said Chromium would not have the NaCl feature working even if other features are working properly.

Hoernchen, 2025-03-13:
btrfs is still affected: I can't mount or boot a rpi400 sd card btrfs that was created 3 years ago with the 6.6.74+rpt-rpi-2712 kernel, due to 
```
[ 7926.089598] BTRFS: device label writable devid 1 transid 167933 /dev/sda2 scanned by mount (134985)
[ 7926.091620] BTRFS info (device sda2): first mount of filesystem e5731234-7dbd-4c96-a83b-48f5e5d7b014
[ 7926.091639] BTRFS info (device sda2): using crc32c (crc32c-generic) checksum algorithm
[ 7926.091654] BTRFS info (device sda2): enabling all of the rescue options
[ 7926.091656] BTRFS info (device sda2): ignoring data csums
[ 7926.091657] BTRFS info (device sda2): ignoring bad roots
[ 7926.091659] BTRFS info (device sda2): disabling log replay at mount time
[ 7926.091661] BTRFS info (device sda2): disk space caching is enabled
[ 7926.091663] BTRFS warning (device sda2): v1 space cache is not supported for page size 16384 with sectorsize 4096
[ 7926.092029] BTRFS error (device sda2): open_ctree failed
```

(this was with "-o ro,rescue=all" which does not help either).

lurch, 2025-03-13:
@popcornmix or @pelwell might be able to confirm, but I wonder if it's the case that a filesystem that was initially created with a 4KB page-size kernel can only later be mounted by a 4KB page-size kernel; and a filesystem that was initially created with a 16KB page-size kernel can only later be mounted by a 16KB page-size kernel? 🤔

Botspot, 2025-03-17:
Signal Desktop
https://github.com/Botspot/pi-apps/issues/2732

popcornmix, 2025-06-27:
Have you reported this to ceph? It is something they will have to fix.
Looks like bug reports are handled through a [mailing list](https://docs.ceph.com/en/latest/rados/troubleshooting/community/).

kovimilu, 2025-06-27:
> Have you reported this to ceph? It is something they will have to fix. Looks like bug reports are handled through a [mailing list](https://docs.ceph.com/en/latest/rados/troubleshooting/community/).

Hi,

No, i have not.
From the things i found, this doesn't seem like a bug to me, do you think this qualifies as a bug?

popcornmix, 2025-07-03:
> From the things i found, this doesn't seem like a bug to me, do you think this qualifies as a bug?

Yes. It is a bug that they are assuming the pagesize is 4K rather than using the appropriate API to query the pagesize
(e.g. `getconf PAGESIZE` from command line, or `sysconf(_SC_PAGESIZE)` from C code). But without knowing their software I don't know where that assumption is made.

There is nothing raspberry pi can do to make their code work on a 16K pagesize kernel (other than provide a manual option to use the 4K pagesize kernel).

kovimilu, 2025-07-05:
> > From the things i found, this doesn't seem like a bug to me, do you think this qualifies as a bug?
> 
> Yes. It is a bug that they are assuming the pagesize is 4K rather than using the appropriate API to query the pagesize (e.g. `getconf PAGESIZE` from command line, or `sysconf(_SC_PAGESIZE)` from C code). But without knowing their software I don't know where that assumption is made.
> 
> It is possible they are depending on another library that makes the 4K assumption, in which case the bug report needs to be punted along to them - but you'd need someone from ceph to diagnose if that is the case.
> 
> There is nothing raspberry pi can do to make their code work on a 16K pagesize kernel (other than provide a manual option to use the 4K pagesize kernel - which is a workaround rather than a fix).

Alright, then i will try to report it as bug in ceph's direction.
By the way, i tried it with a 4KB Kernel, but got the same error message.

this is the error message I get with both kernel page sizes, maybe it it will help go in the right direction:

```
root@ceph-master-1-02:~# ceph orch daemon add osd ceph-master-1-02:/dev/nvme0n1 lvm
Error EINVAL: Traceback (most recent call last):
  File "/usr/share/ceph/mgr/mgr_module.py", line 1809, in _handle_command
    return self.handle_command(inbuf, cmd)
  File "/usr/share/ceph/mgr/orchestrator/_interface.py", line 183, in handle_command
    return dispatch[cmd['prefix']].call(self, cmd, inbuf)
  File "/usr/share/ceph/mgr/mgr_module.py", line 474, in call
    return self.func(mgr, **kwargs)
  File "/usr/share/ceph/mgr/orchestrator/_interface.py", line 119, in <lambda>
    wrapper_copy = lambda *l_args, **l_kwargs: wrapper(*l_args, **l_kwargs)  # noqa: E731
  File "/usr/share/ceph/mgr/orchestrator/_interface.py", line 108, in wrapper
    return func(*args, **kwargs)
  File "/usr/share/ceph/mgr/orchestrator/module.py", line 1100, in _daemon_add_osd
    raise_if_exception(completion)
  File "/usr/share/ceph/mgr/orchestrator/_interface.py", line 240, in raise_if_exception
    raise e
RuntimeError: cephadm exited with an error code: 1, stderr:Inferring config /var/lib/ceph/105042a8-6324-11ef-bae2-d83addf4a913/mon.ceph-master-1-02/config
Non-zero exit code 1 from /usr/bin/podman run --rm --ipc=host --stop-signal=SIGTERM --net=host --entrypoint /usr/sbin/ceph-volume --privileged --group-add=disk --init -e CONTAINER_IMAGE                                     =quay.io/ceph/ceph@sha256:259cbcf17122953765b8febf2e01682caf11c883bd41ffcea12d2afb88f5b9b5 -e NODE_NAME=ceph-master-1-02 -e CEPH_USE_RANDOM_NONCE=1 -e CEPH_VOLUME_OSDSPEC_AFFINITY=None                                      -e CEPH_VOLUME_SKIP_RESTORECON=yes -e CEPH_VOLUME_DEBUG=1 -v /var/run/ceph/105042a8-6324-11ef-bae2-d83addf4a913:/var/run/ceph:z -v /var/log/ceph/105042a8-6324-11ef-bae2-d83addf4a913:/va                                     r/log/ceph:z -v /var/lib/ceph/105042a8-6324-11ef-bae2-d83addf4a913/crash:/var/lib/ceph/crash:z -v /run/systemd/journal:/run/systemd/journal -v /dev:/dev -v /run/udev:/run/udev -v /sys:/                                     sys -v /run/lvm:/run/lvm -v /run/lock/lvm:/run/lock/lvm -v /:/rootfs -v /etc/hosts:/etc/hosts:ro -v /tmp/ceph-tmp4l5ek_p1:/etc/ceph/ceph.conf:z -v /tmp/ceph-tmpe5a27byv:/var/lib/ceph/bo                                     otstrap-osd/ceph.keyring:z quay.io/ceph/ceph@sha256:259cbcf17122953765b8febf2e01682caf11c883bd41ffcea12d2afb88f5b9b5 lvm batch --no-auto /dev/nvme0n1 --yes --no-systemd
/usr/bin/podman: stderr --> passed data devices: 1 physical, 0 LVM
/usr/bin/podman: stderr --> relative data size: 1.0
/usr/bin/podman: stderr Running command: /usr/bin/ceph-authtool --gen-print-key
/usr/bin/podman: stderr Running command: /usr/bin/ceph --cluster ceph --name client.bootstrap-osd --keyring /var/lib/ceph/bootstrap-osd/ceph.keyring -i - osd new a94b22d3-2fde-48b8-b5d5                                     -0d7d6d89ac95
/usr/bin/podman: stderr Running command: nsenter --mount=/rootfs/proc/1/ns/mnt --ipc=/rootfs/proc/1/ns/ipc --net=/rootfs/proc/1/ns/net --uts=/rootfs/proc/1/ns/uts /sbin/vgcreate --force                                      --yes ceph-4c30d0c2-eb2c-4da2-b056-46f26236de8c /dev/nvme0n1
/usr/bin/podman: stderr  stdout: Physical volume "/dev/nvme0n1" successfully created.
/usr/bin/podman: stderr  stdout: Volume group "ceph-4c30d0c2-eb2c-4da2-b056-46f26236de8c" successfully created
/usr/bin/podman: stderr Running command: nsenter --mount=/rootfs/proc/1/ns/mnt --ipc=/rootfs/proc/1/ns/ipc --net=/rootfs/proc/1/ns/net --uts=/rootfs/proc/1/ns/uts /sbin/lvcreate --yes -                                     l 238467 -n osd-block-a94b22d3-2fde-48b8-b5d5-0d7d6d89ac95 ceph-4c30d0c2-eb2c-4da2-b056-46f26236de8c
/usr/bin/podman: stderr  stdout: Logical volume "osd-block-a94b22d3-2fde-48b8-b5d5-0d7d6d89ac95" created.
/usr/bin/podman: stderr Running command: /usr/bin/ceph-authtool --gen-print-key
/usr/bin/podman: stderr Running command: /usr/bin/mount -t tmpfs tmpfs /var/lib/ceph/osd/ceph-0
/usr/bin/podman: stderr Running command: /usr/bin/chown -h ceph:ceph /dev/ceph-4c30d0c2-eb2c-4da2-b056-46f26236de8c/osd-block-a94b22d3-2fde-48b8-b5d5-0d7d6d89ac95
/usr/bin/podman: stderr Running command: /usr/bin/chown -R ceph:ceph /dev/dm-0
/usr/bin/podman: stderr Running command: /usr/bin/ln -s /dev/ceph-4c30d0c2-eb2c-4da2-b056-46f26236de8c/osd-block-a94b22d3-2fde-48b8-b5d5-0d7d6d89ac95 /var/lib/ceph/osd/ceph-0/block
[... cut ...] ev -v /s                                     ys:/sys -v /run/lvm:/run/lvm -v /run/lock/lvm:/run/lock/lvm -v /:/rootfs -v /etc/hosts:/etc/hosts:ro -v /tmp/ceph-tmp4l5ek_p1:/etc/ceph/ceph.conf:z -v /tmp/ceph-tmpe5a27byv:/var/lib/cep                                     h/bootstrap-osd/ceph.keyring:z quay.io/ceph/ceph@sha256:259cbcf17122953765b8febf2e01682caf11c883bd41ffcea12d2afb88f5b9b5 lvm batch --no-auto /dev/nvme0n1 --yes --no-systemd
```
but i will start gathering the things for the bug report