raspberrypi/bookworm-feedback #108: Wayland with WayVNC crashes with no proper recovering

JohnFarl, 2023-10-22:
Problem description:
I use Raspberry mostly with VNC and I'm facing frequent crashes of desktop environment that cause VNC disconnection and loss of control.
If I try to power on HDMI display after these crashes, I see randomly one of these situation:
- black screen
- totally  frozen display, not even mouse cursor moves
- a Raspberry login menu, that is weird considering that I use it with autologin enabled and so it should login automatically, in addition such login menu cannot be used from VNC because at this point VNC connection is already unusable till next OS reboot.

Expected result:
- A more stable environment to avoid crashes in first place that are too frequent despite a clean Raspberry Pi OS image setup.
- Better crash handling with automatic recover of desktop environment and VNC control if for some reason a crash occur.

Device: Pi4 8GB

any1, 2023-10-22:
Might `journalctl` shed some light on what's actually happening?

Although, in case wayvnc is crashing, I'm not sure if the logs would be very useful given that Pi OS is still running an old version. There are some bugs that have been fixed since the previous release.

JohnFarl, 2023-10-22:
These are the anomalies I have in log that could be related to the issue

```
pipewire[35788]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
pipewire[35788]: spa.v4l2: '/dev/video15' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
pipewire[35788]: spa.v4l2: '/dev/video21' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
pipewire[35788]: spa.v4l2: '/dev/video22' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
pipewire[35788]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device

kernel: vc4-drm gpu: swiotlb buffer is full (sz: 389120 bytes), total 32768 (slots), used 27 (slots)
kernel: vc4-drm gpu: swiotlb buffer is full (sz: 442368 bytes), total 32768 (slots), used 1 (slots)
kernel: vc4-drm gpu: swiotlb buffer is full (sz: 561152 bytes), total 32768 (slots), used 7 (slots)
kernel: vc4-drm gpu: swiotlb buffer is full (sz: 274432 bytes), total 32768 (slots), used 5 (slots)
kernel: vc4-drm gpu: swiotlb buffer is full (sz: 565248 bytes), total 32768 (slots), used 5 (slots)
```

hrafnkelle, 2023-11-01:
A "reliable" way for me to reproduce this VNC crash is to run wsjtx for a while. VNC  will usually crash within a 10 minutes and no way to reconnect to the RPi.

any1, 2023-11-02:
There's a chance that this problem might be resolved in the newest version. So, if you're tired of waiting for it to be packaged, you could compile it from source. If that doesn't fix the problem, the stdout and stderr output from wayvnc along with a backtrace would be very useful.

andrum993, 2023-11-02:
Ideally the wayvnc service unit file would configure systemd so it attempted to restart wayvnc if it detects the wayvnc server has crashed, rather than relying on a user going in and mamually rstarting it.

any1, 2023-11-02:
> Ideally the wayvnc service unit file would configure systemd so it attempted to restart wayvnc if it detects the wayvnc server has crashed, rather than relying on a user going in and manually rstarting it.

Yes, this would be a more robust way of running it than xdg-autostart. It could even be started as a systemd user service.

The best way would, of course, be to modify wayvnc so that it can run detached from the compositor and start it as a system service. Then, when a compositor comes along, it would attach to the compositor. I've experimented with this a bit; just shuffling around some code and not taking into account stability or any kind of resource management, I was able to get it working within quite a short time-frame. However, finishing this is pretty low on my list of priorities.

andrum993, 2023-11-02:
I hadn't looked at how wayvnc was being managed at present - I was just assuming it was getting run as some sort of systemd service - sounds like that's not the case.

andrum993, 2023-11-02:
And RealVNC will be coming along at some point with their Wayland-compatible VNC server, so that will probably end up going back in as the default VNC server on Raspberry Pi OS.

hrafnkelle, 2023-11-03:
I hooked up a monitor to my rpi and reproduced the issue. I don't think the issue is in wayvnc. The whole desktop environment goes down and respawns. VNC is not activated again until after I re-log into the desktop (auto login only seems to work on startup?)