raspberrypi/bookworm-feedback #109: sudo apt update reports DEPRECATION

sgalke, 2023-10-22:
"W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details"

system: bookworm 32-bit / Pi 4B

bug? irrelevant??

Rom1deTroyes, 2023-10-22:
Just moving `/etc/apt/trusted.gpg` to `/etc/apt/trusted.gpg.d/trusted.gpg` does the trick, but it's surely not a best practice ?
(I have the same problem, using a fresh installation with `2023-10-10-raspios-bookworm-armhf-lite.img.xz` image).

sgalke, 2023-10-22:
thanks. 
i really hate this security scrub - Sicherheitsgestrüpp - Gommage de sécurité

DrOuille9, 2024-08-12:
Current **2024-07-04-raspios-bookworm-armhf-lite.img** does the same spit out message after a **fresh install** on a RPi 2B...
August 2024

As root :
```
apt update
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
28 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```
It can disappointing after a fresh and clean install ?

lurch, 2024-08-29:
> It can disappointing after a fresh and clean install ?

It's only a harmless warning, and can be safely ignored.

DrOuille9, 2024-08-29:
> > It can disappointing after a fresh and clean install ?
> 
> It's only a harmless warning, and can be safely ignored.

A warning message can give stress to usual users, everybody is not a dev. It's like a little light on a car control panel, if it's here, there's for sure a problem... ;)

plugwash, 2024-08-30:
It's not about the format of the key itself, but about how it's installed. I need to get around to sorting this but it's something that requires care.

XECDesign, 2024-09-26:
Had to delete a xenophobic comment that claimed that this issue prevents the internet from working. It does not, it's just a message. Everything still behaves exactly the same way.