raspberrypi/bookworm-feedback #111: CubicSDR

rpibkm, 2023-10-22:
Hello. I have a Raspberry Pi 4B with 8 GB RAM. I installed bookworm, but is has several issues. One of them is that the program CubicSDR (software-defined radio). 1) it takes very long to find the Device (the SDR dongle via USB). 2) mouse movements are choppy. 3) it crashes after a few minutes of use. I hope you can solve this issue and not remove CubicSDR from the repository. Thanks.

qrp73, 2023-10-24:
> 1) it takes very long to find the Device (the SDR dongle via USB)

I think this is due to bug in USB kernel drivers, it was fixed but still missing from release image and updates. See issue  #83 

You can fix it with direct apply for pull request with fix for this bug:
```
sudo rpi-update pulls/5642
```

As alternative, you can update linux kernel to 6.5 or 6.6, it already has fix for this bug:
```
sudo rpi-update rpi-6.6.y
```

But note that update kernel may break some software... 
I'm using it and it works ok, but who knows what can happens on your system

qrp73, 2023-10-24:
> 2) mouse movements are choppy

I think this is due to issue #65 
Unfortunately there is no fix for that...

rpibkm, 2023-10-24:
okay, thank you for your updates. Have a great day!

closed 2023-10-24 by rpibkm