raspberrypi/bookworm-feedback #112: Wi-Fi issues

rpibkm, 2023-10-22:
Booworm has a few issues with Wi-Fi under Raspberry Pi 4B (8 GB RAM).
1) Wi-Fi Hotspot is created easily and devices (iPad, MacBook Pro) can connect to it, but it does not give internet connectivity. And yes, I have an active ethernet connection. Hotspot with Bullseye works very well, but not with Bookworm. 
2) Wi-Fi Security: WPA3 does not work. WPA & WPA2 Personal works but it is detected as WEP (weak security).

aallan, 2023-10-23:
See https://www.raspberrypi.com/documentation/computers/configuration.html#host-a-wireless-network-on-your-raspberry-pi

rpibkm, 2023-10-24:
okay, thank you for your reply and link for further reading.

lurch, 2023-10-25:
I've just tested this myself, and everything seems to work as expected...

Pi 3B+ running Bookworm 32-bit connected via ethernet to the local network (verified that the Pi 3B+ can access the internet). Pi 4B running Bookworm 64-bit with no ethernet connected.

On the Pi 3B+ I created a Wireless Hotspot, with "WPA & WPA2 Personal" security. On the Pi 4B I connected to this wireless hotspot, and then Pi 4B then has no problems at all accessing the internet. The "Connection Information..." window on the Pi 4B also shows that the hotspot is using "WPA/WPA2" security.

So I'm going to close this issue as "can't reproduce", but please feel free to provide more information if you're still having problems.

closed 2023-10-25 by lurch

rpibkm, 2023-10-25:
Hi Lurch, thank you for your feedback. If I read your comments correctly, you created a wireless hotspot in Pi 3B+ and tested it in Pi 4B and everything worked fine. That is great. Would it be possible for you to test the reverse also? Creating the wireless hotspot in Pi 4B and testing it in another Pi? In my case, as I described before, I used a Pi 4B running Bookworm, and the wireless hotspot does not provide internet to my iPad and MacBook Pro. And the WPA/WPA2 security is detected as WEP. Thank you, have a great day. As a footnote, my wireless hotspot in Pi 4B running Bullseye works very well giving internet to my other devices. But still WPA/WPA2 security is detected as WEP security.

lurch, 2023-10-25:
> Would it be possible for you to test the reverse also? Creating the wireless hotspot in Pi 4B and testing it in another Pi?

I just tested this by swapping over the SD cards and ethernet cable, and as expected that configuration works fine too.

Not really sure what else I can suggest, but you might be able to get more detailed help from https://forums.raspberrypi.com/

rpibkm, 2023-10-25:
Okay, thank you very much for testing again and for your suggestion to get more help from the RPi Forums.
Take care. Rick.