raspberrypi/bookworm-feedback #114: 'raspi-config nonint do_serial' ignores nonint

s-k-tttttttt, 2023-10-22:
The version of raspi-config that shipped with the original bookworm release worked, and an update was released last week (raspi-config all 20231018).

Instead of just applying the setting silently when issuing '**raspi-config nonint do_serial 1**', the blue 'Would you like a login shell to be accessible over serial?' question appears and waits for input.

After the input is manually entered, a second question 'Would you like the serial port hardware to be enabled?' appears and waits for input.

Then (after having answered twice) a third blue screen appears telling you that the serial login shell and serial interface have been set as requested, again waiting for user input.

Have seen this behaviour on 64-bit and 32-bit Lite versions, on Pi4B, Pi3B and Pi2ZeroW and assume happens across the board.

spl237, 2023-10-23:
The syntax of that command has been changed to make it compatible with Pi 5, and it should no longer be used in nonint mode.

For noninteractive setting of serial console and hardware, use the new functions do_serial_hw and do_serial_cons.

s-k-tttttttt, 2023-10-23:
Thankyou kindly for this information.

As of this morning as I write this, google is not finding anything with regards to either do_serial_cons or do_serial_hw. So I have fumbled around for a little trying understand these.

I discovered that do_serial_cons and do_serial_hw functions have been added into the raspi-config programme which is a shell script located at /usr/bin/raspi-config

However as far as I can tell, the only way these functions can be called is **AFTER** the shell script calls 'whiptail' which is the programme raspi-config uses to bring up the blue interactive dialogue screens.

Please forgive me if I am wrong on this as I am relatively new to trying to decipher and use raspi-config, but it does seem there is a bug in the recent raspi-config changes... In order to support non-interactive configuration of other options, calls to whiptail are preceded by:
`if [ "$INTERACTIVE" = True ]; then`
where as that test is missing within the do_serial() function. Therefore from my understanding of shell script, I cannot see how do_serial() will allow non-interactive use. Again - sorry if I am missing something.

As an alternative, is there any other way that the serial console can be configured non-interactively if raspi-config no longer supports this non-interactively?

spl237, 2023-10-23:
Simply call those functions as if you were calling do_serial noninteractively - "sudo raspi-config nonint do_serial_hw 0" (or 1 to disable the serial hardware), and "sudo raspi-config nonint do_serial_cons 0" (or 1 to disable the serial console). No need for whiptail or anything else in interactive mode. Those two functions are specifically designed for non-interactive use, so do not require the if $INTERACTIVE test; just as do_serial itself is specifically designed for interactive use and hence also does not require the test.

do_serial cannot be used for non-interactive use any more; it is purely for use when using raspi-config interactively. do_serial_hw and do_serial_cons provide exactly the same functionality, and are the way to achieve the same effect non-interactively. Does that help?

spl237, 2023-10-23:
Or to put it another way.

If before you were doing
raspi-config nonint do_serial 0  (to enable both hardware and console)
you now do
raspi-config nonint do_serial_hw 0
raspi-config nonint do_serial_cons 0

If before you were doing
raspi-config nonint do_serial 1 (to disable both hardware and console)
you now do
raspi-config nonint do_serial_hw 1
raspi-config nonint do_serial_cons 1

If before you were doing
raspi-config nonint do_serial 2 (to enable hardware and disable console)
you now do
raspi-config nonint do_serial_hw 0
raspi-config nonint do_serial_cons 1

This is actually more logical, and makes it more obvious what each setting actually does.

s-k-tttttttt, 2023-10-23:
Thanks again kindly. Your explanation makes sense and is clear.

lurch, 2023-10-24:
ping @aallan as this might require updates to https://www.raspberrypi.com/documentation/computers/configuration.html#the-raspi-config-command-line-interface

s-martin, 2024-11-03:
> Or to put it another way.
>
> If before you were doing
> raspi-config nonint do_serial 0  (to enable both hardware and console)
> you now do
> raspi-config nonint do_serial_hw 0
> raspi-config nonint [...]

Just a follow up question:
Is this backwards compatible?

Does this work also with bullseye?

spl237, 2024-11-03:
> Just a follow up question: Is this backwards compatible?
> 
> Does this work also with bullseye?

Bullseye will have an older version of raspi-config anyway, so it won't have changed from the previous behaviour.