raspberrypi/bookworm-feedback #115: bluetooth hci0: firmware: failed to load

JohnFarl, 2023-10-23:
I have these errors related to Bluetooth in journal 

```
kernel: bluetooth hci0: firmware: failed to load brcm/BCM4345C0.raspberrypi,4-model-b.hcd (-2)
kernel: firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
bluetoothd[817]: src/plugin.c:plugin_init() Failed to init vcp plugin
bluetoothd[817]: src/plugin.c:plugin_init() Failed to init mcp plugin
bluetoothd[817]: src/plugin.c:plugin_init() Failed to init bap plugin
bluetoothd[817]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
bluetoothd[817]: sap-server: Operation not permitted (1)
bluetoothd[817]: Failed to set privacy: Rejected (0x0b)
bluetoothd[817]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager obj>
```

Device: Raspberry Pi 4 8GB

pelwell, 2023-10-23:
Does Bluetooth work?

JohnFarl, 2023-10-23:
> Does Bluetooth work?

Bluetooth mouse seems working, I don't know if  also other bluetooth features and mode work. But these are error level log entries not just warnings, so something needs to be fixed.

pelwell, 2023-10-23:
What needs to be fixed is the kernel displaying error messages when it ends up using the unqualified name to load an appropriate firmware - it's not an error.

closed 2023-10-23 by pelwell