raspberrypi/bookworm-feedback #116: SSH via ethernet doesn't work

llmsrhl, 2023-10-24:
**Hardware**: Raspberry Pi 4 Model B 4gb ram
**OS**: Raspberry Pi OS Bookworm (64-bit)

**Issue**
I installed a 64-bit bookworm version of the OS on my Raspberry 4
Via WI-FI I can connect from my MacBook to the Raspberry via ssh(like ssh user@ssh user@raspberrypi.local), but as soon as I turn off the wi-fi on the Raspberry and connect the Ethernet cable, I can't connect
On the 32-bit bullseye I was able to easily connect via cable
**Additional info** [raspi forum](https://forums.raspberrypi.com/viewtopic.php?t=358318)

qrp73, 2023-10-25:
What shows the following command?
```
ip addr
```

ghollingworth, 2023-10-25:
Works as expected, you need to have a DHCP server to give your network adapter an IP address because that's the way ethernet works.  It's possible you can give it a static IP address, which will bring up the network with the DHCP, but otherwise this is the way it is supposed to work.

closed 2023-10-25 by ghollingworth

lurch, 2023-10-25:
I think this might be the same problem as #104

llmsrhl, 2023-10-25:
> What shows the following command?
> 
> ```
> ip addr
> ```
> 
> try to connect to ip address that it shows for eth0

I have attached images on the forum [you can see](https://forums.raspberrypi.com/viewtopic.php?t=358318)