raspberrypi/bookworm-feedback #117: vnc activation not persistent (bookworm 64-bit, wayland, , RPi 4B)

sgalke, 2023-10-26:
previously enabled vnc server (wayvnc) disables itself after some time ...

re- enabling it via raspi-config shows:

Warning: ../src/ctl-server.c: 703: Deleting stale control socket path "/run/user/1000/wayvncctl"

bug?

lurch, 2023-10-26:
Perhaps similar to #108 ?

sgalke, 2023-10-26:
i dont think so. no crashes here, even no use of the remote rpi for a while... then trying to conect: refused (tiger- and real-vnc, no difference)  :(( 
i have to login via ssh and re-enable vnc w/ raspi-config and besides the Warning message can connect with  real-vnc or tiger-vnc.

Jhutjens92, 2023-10-27:
This happening here aswell. Every morning we have to re-enable the vnc interface using raspi-config.

spl237, 2023-10-27:
We have a new version of wayvnc waiting to be released - it is possible that this will fix the problem; otherwise it will need to be reported to the author of wayvnc.

any1, 2023-10-30:
> We have a new version of wayvnc waiting to be released

Just to be sure, you're not waiting on me, are you? I released almost 4 weeks ago. Did you miss my email?

any1, 2023-10-30:
```
Warning: ../src/ctl-server.c: 703: Deleting stale control socket path "/run/user/1000/wayvncctl"
```

This happens if wayvnc did not exit cleanly from a previous run. It can happen as a result of a crash or because it was killed.

dts350z, 2023-11-03:
Same issue here on PI 4 and 5. Very frustrating. Sometimes takes multiple config change plus reboot, to get to work for a while, then repeat.

0evmc, 2023-11-03:
I am having the same or a very similar issue. I do not currently have a microHDMI adapter, so I was just going to use SSH to enable VNC in the meantime, but I have yet to get VNC to accept an incoming connection, and I've seen the Warning mentioned in the post and the screenshot shown in Jhutjens92's comment. I'm unsure what to do from here.

spl237, 2023-11-03:
The new version of wayvnc should be in apt early next week - hopefully this will solve some of these issues.

dts350z, 2023-11-03:
I think that screen shot Jhutjens92 is mixing apples and oranges (WylandVNC vs. X11vnc) but perhaps points to part of the problem in the apparent state confusion.

Using ps -ef | grep VNC I see:

[image]

sudo -u glenn XDG_RUNTIME_DIR=/run/user/1000 WAYLAND_DISPLAY=wayland-1 wayvnc --render-cursor --keyboard=us is what you see only when you have just enabled the VNC interface in the Raspbery Pi Config GUI, but haven't yet rebooted.

wayvnc --render-cursor --keyboard=us is what you see (before or after reboot) when the service is actually running.

I also suspect there are two different failure mechanisms, one which is periodic in nature, and one which happens anytime the server crashes due to an error.

Various confusions, in the OS,  over waylandVNC vs Xll VNC being active may also be playing a part?

dts350z, 2023-11-03:
Re a new version of wayvnc, I hope someone is stress testing, and including at 4K screen resolution. I suspect it is less stable at that resolution vs. 1920x1080, etc.

dts350z, 2023-11-07:
Now on wayvnc 0.7.1-1~bpo12+rpt2 (version updated yesterday). The same issue still exists.

0evmc, 2023-11-07:
> Now on wayvnc 0.7.1-1~bpo12+rpt2 (version updated yesterday). The same issue still exists.

Good to know. I'm just doing everything I can via SSH for right now and just not using VNC at all.

m1k14U, 2023-11-07:
all of this is what I was looking for to explain what happened to my Rasp Pi4B, I was frustrated because I thought I put something wrong or erased something important

sgalke, 2023-11-10:
The "Screen Blanking" setting seems to have an influence: when turned off, the VNC setting is retained, but when turned on, the error occurs after a while. 

By the way, the "screen blanking" setting has no effect on the monitor: it is always on! 

(bookworm 64-bit, wayland, RPi 4B with bluetooth keyboard & hdmi monitor)

dts350z, 2023-11-10:
Screen blanking is off by default here. Two systems pi 4b and pi 5. VNC error and disabling in raspberry pi config continues. This is after this week's updates to wayvnc and raspberry pi config.

vasilvestre, 2023-11-12:
It also happens to me when I unplug my HDMI cable.

iohe, 2023-11-19:
rpi 4b, 4g ram, hdmi monitor 1920x1080.
default bookworm, up 2 date 
Way to reproduce: 
    connect via vnc with RealVnc 6.21.406
     watch 2-3 seconds.
    disconnect.
    Repeat. sigsegv will appear, maybe at 6th try or 4th..ymmv  

rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT], 8) = 0
rt_sigprocmask(SIG_SETMASK, [INT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT], 8) = 0
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\22\0\0\0\1\0\f\0\0\0\0\0\22\0\0\0\4\0\30\0\0\0\0\0\0\0\0\0\200\7\0\0"..., iov_len=1308}, {iov_base="", iov_len=2788}], msg_iovlen=2, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_DONTWAIT|MSG_CMSG_CLOEXEC) = 64
sendmsg(11, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\4*\32\204+&\312\341nU\322\35\313\316\331j4\tT;[", iov_len=22}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL) = 22
epoll_ctl(4, EPOLL_CTL_MOD, 11, {events=EPOLLIN, data={u32=2111292368, u64=367183512528}}) = 0
sendmsg(11, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base=" \0\260F\330cV\310FxU\200|\333eB\211\326\rk\202\232\201Y\250G\20?t\305@U"..., iov_len=1838421}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL) = -1 ECONNRESET (Connection reset by peer)
rt_sigprocmask(SIG_SETMASK, [INT], NULL, 8) = 0
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\22\0\0\0\1\0\10\0\10\0\0\0\0\0\24\0\23\0\0\0\1\0\0\0\f\0\0\0", iov_len=28}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_DONTWAIT|MSG_NOSIGNAL) = 28
epoll_pwait(4, [{events=EPOLLIN, data={u32=2111140704, u64=367183360864}}, {events=EPOLLIN|EPOLLHUP, data={u32=2111292368, u64=367183512528}}], 16, -1, NULL, 8) = 2
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT], 8) = 0
rt_sigprocmask(SIG_SETMASK, [INT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT], 8) = 0
rt_sigprocmask(SIG_SETMASK, [INT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT], 8) = 0
read(6, "\1", 256)                      = 1
read(6, 0x7ff74c29b8, 256)              = -1 EAGAIN (Resource temporarily unavailable)
read(11, "", 8192)                      = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x18} ---
+++ killed by SIGSEGV +++
pi@raspberrypi:~ $

Thread 1 "wayvnc" received signal SIGSEGV, Segmentation fault.
0x0000007fb522d450 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
(gdb) bt
#0  0x0000007fb522d450 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#1  0x0000007fb522a50c in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#2  0x0000007fb522dba0 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#3  0x0000007fb522e1a0 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#4  0x0000007fb522dbe4 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#5  0x0000007fb522dcc4 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#6  0x0000007fb52391d8 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#7  0x0000007fb5228d50 in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#8  0x0000007fb522e09c in ?? () from /lib/aarch64-linux-gnu/libneatvnc.so.0
#9  0x0000007fb5312d4c in aml_dispatch () from /lib/aarch64-linux-gnu/libaml.so.0
#10 0x00000055767185d0 in ?? ()
#11 0x0000007fb4fd7780 in __libc_start_call_main (main=main@entry=0x5576717d80, argc=argc@entry=3, argv=argv@entry=0x7ffa57eae8)
    at ../sysdeps/nptl/libc_start_call_main.h:58
#12 0x0000007fb4fd7858 in __libc_start_main_impl (main=0x5576717d80, argc=3, argv=0x7ffa57eae8, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:360
#13 0x0000005576718bf0 in ?? ()

pi@raspberrypi:~ $ wayvnc -V
wayvnc: 0.7.2
neatvnc: 0.7.1
aml: 0.3.0

pi@raspberrypi:~ $ apt list wayvnc -a
Listing... Done
wayvnc/stable,now 0.7.1-1~bpo12+rpt2 arm64 [installed,automatic]

sudo apt install wayvnc
wayvnc is already the newest version (0.7.1-1~bpo12+rpt2).

any1, 2023-11-19:
That backtrace would be more useful with debug symbols.

m1k14U, 2023-11-27:
somebody knows news about the problem ???

any1, 2023-11-27:
> somebody knows news about the problem ???

Well, there's this:  https://github.com/any1/wayvnc/pull/278

sgalke, 2023-11-27:
seems to be stable now 2023-Nov-27 after some system updates.

sgalke, 2023-11-27:
perhaps someone will close this issue

popcornmix, 2023-11-27:
@sgalke you as the author is the issue can close it.

sgalke, 2023-11-27:
ok