raspberrypi/bookworm-feedback #119: WayFire stays on GB keyboard

villeneuve, 2023-10-26:
There is a minor bug with non GB keyboard
Something similar to [issue 60](https://github.com/raspberrypi/bookworm-feedback/issues/60) 

In the WayFire config
```
<super> KEY_Q close a window
and
<ctrl><alt>KEY_W open netmanager
```

However with my french keyboard (AZERTY instead of QWERTY). And the keyboard correctly to set to FR in raspi-config:
I must  press
<super>  and **KEY_A** to close a window
and
<ctrl><alt> and **KEY_Z** to open netmanager

Otherwise since issue 60 has been closed everything is working OK with the keyboard set to FR.

Thank you

qrp73, 2023-10-27:
There are 3 config files for keyboard layoyt.

This is an example with 3 layouts: US, RU, UA and Alt+Shift key for layout switch. 

1) for console: sudo nano /etc/default/keyboard
```
XKBMODEL="pc105"
XKBLAYOUT="us,ru,ua"
XKBVARIANT=""
XKBOPTIONS="grp:alt_shift_toggle"
```

2) for Wayland: nano ~/.config/wayfire.ini
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru,ua
xkb_variant =
```

3) for login screen: sudo nano /usr/share/greeter.ini
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru,ua
xkb_variant =
```

If you want English and French layout, just replace "us,ru,ua" with "us,fr".

spl237, 2023-10-27:
I'm not sure that is fixable - the key codes used in the wayfire config file are the hardware key names, not the translated names. It's only really a "bug" in the sense that the config file has the "wrong" names - the actual keys pressed are correct.

Closing on the grounds that I don't think this is possible to fix.

closed 2023-10-27 by spl237

villeneuve, 2023-10-27:
@spl237 Ok understand. Fair enough. However I believe that WayFire shouldn't talk directly to the hardware but instead talk to the system (Linux) which is doing key translation. Anyway it's minor. Thank you

@qrp73 Thank you for that. We already discussed this on issue https://github.com/raspberrypi/bookworm-feedback/issues/60. However in this particular problem this isn't the solution. The settings in these 3 files are already set to FR on my system. I don't want to set my keyboard to US and FR. I want FR only (same on all my Linux computers for more than 20 years). The (stupid) difference between UK/US/GB/Many others and FR is the key location. A french keyboard is AZERTY not QWERTY  (most of the world keyboards are QWERTY). When you switch your keyboard from cyrillic to latin the first letter you have just at the right of the TAB key is "Q" the next is "W". On my keyboard it's "A" and "Z". It's printed on the keys like on every computer sold in France. My raspberry pi400 has an AZERTY (french) keyboard. Anyway this closed. Дякую і доброго дня

qrp73, 2023-10-27:
@villeneuve, did you tried to set ```xkb_variant=azerty```?