raspberrypi/bookworm-feedback #122: Enabling the readonly file system overlay disables desktop autologin, and prevents user from logging in.

JinShil, 2023-10-27:
Steps to reproduce:
1.  Enable desktop autologin and verify that it works.
2.  Enable the readonly file system overlay, and reboot
3. Notice that, after rebooting, the user is no longer autologged into the desktop, and instead is presented with a login dialog window.
4. Further notice that after entering the password, it appears the user is logged in, but then immediately logged out.

lurch, 2023-10-27:
Ahhh, possibly the same thing as #22 but just occurring in a slightly different scenario?
(i.e. some part of the "login process" is trying to write to disk, but failing, which causes the login process itself to fail?)

JinShil, 2023-11-09:
I can no longer reproduce this symptom.  I'm not sure what changed, but it appears to be fixed.  Closing.

closed 2023-11-09 by JinShil