raspberrypi/bookworm-feedback #123: Locale is non-UTF-8 in bookworm images causing some applications to fail

theofficialgman, 2023-10-29:
Locale is non-UTF-8 causing some applications to fail to launch.

https://forums.raspberrypi.com/viewtopic.php?t=357707
https://github.com/davidk/PrusaSlicer-ARM.AppImage/issues/29

Issue observed on Bookworm ARM64 image (October 10th 2023). May be present on ARMhf images as well but I have not tested.

ghollingworth, 2023-10-29:
Can you actually define what the issue is?  The discussion on the forums doesn’t say what the actual issue is or where it is wrong.  As far as we know the issue is one in Debian

pelwell, 2023-10-29:
As an observer of the issue I would guess that /xdg/locale.gen contains all of the possible locales, including UTF-8 and ISO-8859-*, and that the locale selection logic is choosing an ISO version where it should (for reasons of compatibility) choose the UTF-8 version instead.

theofficialgman, 2023-10-29:
the issue is as described here -> https://forums.raspberrypi.com/viewtopic.php?t=357707#p2144544 . the issue simply put is that `.UTF-8` is not added to the end of the language name as it should be.
I have verified this happens on both the ARMhf and ARM64 Bookworm images.

my assumption is the bug is in your initial setup GUI script (piwiz https://github.com/raspberrypi-ui/piwiz) and the language selection.

a quick search of that codebase shows you use raspi-config to set the locale based on user supplied input in the gui:
https://github.com/raspberrypi-ui/piwiz/blob/963cfad174151c051e3b9131443ef13de126d605/src/piwiz.c#L788-L811
https://github.com/raspberrypi-ui/piwiz/blob/963cfad174151c051e3b9131443ef13de126d605/src/piwiz.c#L710-L736

your code does attempt to add the extension `.UTF-8` to the end of the name if there is no extension present but this appears to not be happening https://github.com/raspberrypi-ui/piwiz/blob/963cfad174151c051e3b9131443ef13de126d605/src/piwiz.c#L876
CC: @spl237

spl237, 2023-10-29:
Looks like I have the sense of that test wrong in the tertiary operator - it is adding a UTF-8 when it already has one and not adding one when it doesn't. Trivial to fix - will do so in next release.

spl237, 2023-10-30:
OK, I think I've found the problem - the extension for UTF-8 was not being added to the call to raspi-config in the wizard. Will be fixed in the next release.

closed 2023-10-30 by spl237