raspberrypi/bookworm-feedback #124: Mathematica requests activation on Pi 5

byzantic, 2023-10-29:
.. whereas on a Pi 400 running Bookworm, everything works OK.

Could this be due to the different graphics driver implementation for Pi 5? Elsewhere on the web, this issue has been caused in the past by a permissions issue for `/dev/vchiq` but I believe this no longer exists in the Pi 5 configuration?

pelwell, 2023-10-29:
As I understand it, Mathematica is using the ability to perform a vchiq operation as a check that it is running on a Raspberry Pi. The absence of vchiq on the Pi 5 must have the same effect as incorrect permissions did on earlier Pis - a failure to authenticate.

byzantic, 2023-10-29:
Which means that one can't simply create an empty file with the correct permissions, since an actual operation needs to be performed..

Are the operations available via another device? In which case, maybe create a link to that device as `/dev/vchiq` ?

Otherwise it sounds as though Mathematica need to make a change, but I suspect that would take some time.

popcornmix, 2023-10-30:
>Are the operations available via another device? In which case, maybe create a link to that device as /dev/vchiq ?

`/dev/vcio` provides a different interface to the firmware which can also support similar features like vcgencmd.
But a symlink wouldn't work - the interface is much more basic.

>Otherwise it sounds as though Mathematica need to make a change, but I suspect that would take some time.

Yes. I understand they have been asked, but I am not sure of the status.

byzantic, 2023-11-01:
Latest version of the Wolfram engine fixes this, so closing the issue.

closed 2023-11-01 by byzantic