raspberrypi/bookworm-feedback #125: volume control and bluetooth speaker under pulse

rbnpi, 2023-10-29:
Volume control can select and control the volume of a bluetooth speaker under pipewire, but can only select the speaker, but NOT control its volume when operation under pulseaudio. It can mute it in both cases.
The volume control on the driving sound source app eg iPlayer on Chromium is effective however.

spl237, 2023-11-23:
This is almost certainly due to the selected profile for the device in question. With some Bluetooth devices, the handset profile and the high-quality audio profile do not both allow the volume to be controlled. Try using the Device Profiles option in the output device select menu to change profile and see if the volume control is active in different profiles.

lurch, 2024-07-23:
@rbnpi Did you try the suggestions given above? Are you still having problems?
I'm tempted to close this issue due to lack of activity.

rbnpi, 2024-07-23:
Never sorted it 100%, but I can live with it. Have other BT devices which work OK. It sometime worked, but once it failed, I had to reinstall the OS to get it to work again. Is there any way to complete reset stored bt profies etc without doing this?
I guess I'm happy ofr you to close, but would be nice to know about effectively factory reset to BT.