raspberrypi/bookworm-feedback #128: Unable to login on Bookworm 64-bit Raspberry Pi4b

sfsutherland-git, 2023-10-31:
New Bookworm Raspberry OS installs fail to login after turning auto login off.

I can reproduce this MASSIVE BASIC OS-SECURITY FAILURE on all of my 4b pi's. Using any imager (PI, rufus, etc) latest versions with downloaded PI BOOKWORM IMAGE IN ANY CONFIGURATION from Raspberryy Pi website.

YOU'VE GOT A TRIPLE-MAJOR FU!!!

After cleanly installing via the guided installers on 4 different USB drives, I reboot with auto login button 'ON'. No problems coming back up to the desktop of BOOKWORM. Both before & after performing updates & rebooting again with auto login button in 'ON', no problem coming back up to the desktop.

At this point. I turn auto login button to 'OFF' close rasp-config & reboot. Upon login desktop screen for sign in, when I enter my written down password I am not able to logon and directed to try again, which just repeats over & over.

Using raspberry pi Bullseye 64-bit, no such issues occur on the same 4b's & USB drives.

Using Ubuntu 64-bit for PI installs & performs without any issues as well on the same hardware.

GET THIS FIXED!!!!
SO IT GD WORKS!!!
MAKE IT HAPPEN FAST!!!

I use the following:
US english. US keyboard. New York, 105 key keyboard, UTF-8. during setup prompts, set my pc name, set my pc user, set my pc user password.

[Steven.F.Sutherland@gmail.com](mailto:Steven.F.Sutherland@gmail.com)

lurch, 2023-11-13:
I think I've found the problem here - I assume you must have a character like `@` or `#` (or similar) in your password which maps to a different key on a US keyboard layout compared to a GB keyboard layout?

When you set the keyboard layout during the first-boot wizard, it sets the keyboard layout both for your logged-in desktop session, and also (separately) for the login-window itself (for when you've turned off auto-login). Unfortunately there was a bug in `raspi-config` (which has since [been fixed](https://github.com/RPi-Distro/raspi-config/commit/a9a8835915a6118687b5b9e22dd2ba11dc96fdac) ) which meant that the keyboard-layout change for the login-window was being made in the wrong file. Therefore, when you disable auto-login and reboot, the login-window is still using the default GB keyboard layout instead of the US keyboard layout that you're expecting, and so your password doesn't get typed in correctly. The workaround for this is to ensure your version of Raspberry Pi OS has been fully updated, and then use the Raspberry Pi Configuration program to (temporarily) change your keyboard to a non-US layout, click OK, then change it back to a US layout, and click OK again. And this will then update the login-window (that you see when auto-login is disabled) to be using the correct keyboard layout, so you should be able to type in your password correctly.

Apologies for the inconvenience, and please let us know if this doesn't fix your problem.

pelwell, 2023-11-13:
[ Complaint masquerading as question deleted ]

lurch, 2023-11-13:
>  Is this raspi-config bug fixed in the BOOKWORM image files one downloads from the PI Website.

No, the Bookworm image files downloadable from https://www.raspberrypi.com/software/operating-systems/ (or via Raspberry Pi Imager) still contain this raspi-config bug, so you'll need to do the workaround I described above _before_ disabling auto-login.

sfsutherland-git, 2023-11-27:
Hi Andrew...your login bug fix worked. Thank you. Steven

rmitchellgit, 2023-11-28:
Andrew,

Are there plans in place to fix this so that you don't have to change the keyboard to non-US and then back to US?

Thanks.

lurch, 2023-11-28:
It has already been fixed in `raspi-config`, which means the workaround won't be needed after we release the next version of Raspberry Pi OS.

lurch, 2023-12-08:
Fix is now included in the [latest Raspberry Pi OS](https://www.raspberrypi.com/news/im-theme-ing-of-a-dark-christmas/).

closed 2023-12-08 by lurch

sfsutherland-git, 2023-12-08:
Thanks for the updated status Andrew.

Steven