raspberrypi/bookworm-feedback #129: Incorrect Processor showing on Raspberry Pi 5

makermelissa, 2023-10-31:
Not sure if this is the correct place to report this, but when typing `cat /proc/cpuinfo` I see the following:
```
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

Hardware	: BCM2835
Revision	: d04170
Serial		: ee14f78c1312ce93
Model		: Raspberry Pi 5 Model B Rev 1.0
```

Under Hardware it lists BCM2835, which was the CPU used on the Raspberry Pi 1 and 2. It should show BCM2712 I believe.

pelwell, 2023-10-31:
The string is hard-coded to `BCM2835` in the downstream modification to the arm64 cpuinfo code - even Pi 4 shows `BCM2835`. The whole concept of machine types is missing in the arm64 world, so aside from parsing the revision value and turning the processor ID into a string, which isn't very useful, there isn't much we can do.

I'm considering dropping it, but there's always the possibility that some random bit of software is looking for it. Any thoughts, @popcornmix ?

pelwell, 2023-10-31:
On the 32-bit kernel the Hardware value is "BCM2711" on Pi 4s and "BCM2835" on the other Pis. On BCM2837 the value is no less correct on 64-bit than it is on 32-bit, but it's still not very useful. So we either expand the downstream patch with a decode of the board revision or we shrink the patch by removing that line. I vote for the latter.

popcornmix, 2023-10-31:
I'm okay with the latter.

makermelissa, 2023-10-31:
I believe RPi.GPIO uses the Hardware field.

andrum993, 2023-10-31:
Just to check: with the downstream patch reverted, on the same hardware, will `/proc/cpuinfo` report the same SoC name in the `Hardware` field on both 32-bit and 64-bit OS? I think if you're going to move away from `Hardware: BCM2835` meaning "This is some kind of Raspberry Pi  - check revision field to see which one", then the SoC name provided in the `Hardware` field should be independent of what the bit depth of the running OS happens to be.

popcornmix, 2023-10-31:
The `Hardware: BCM2835` line will disappear on arm64 kernel.
This matches the upstream kernel behaviour.

andrum993, 2023-10-31:
> The `Hardware: BCM2835` line will disappear on arm64 kernel. This matches the upstream kernel behaviour.

I see. So the `Hardware` line will only appear on 32-bit, and will display the following:

BCM2708/2835-based Pis: `Hardware: BCM2835`
BCM2709/2836-based Pis: `Hardware BCM2835`
BCM2710/2837-based Pis: `Hardware BCM2835`
BCM2711-based Pis: `Hardware: BCM2711`
BCM2712-based Pis: `Hardware: BCM2712`

Is that correct?

pelwell, 2023-10-31:
Yes, except for the past line b

pelwell, 2023-10-31:
Yes, except for the he last line - 2712 can't run a 32-bit kernel.

closed 2023-10-31 by pelwell

andrum993, 2023-10-31:
> Yes, except for the he last line - 2712 can't run a 32-bit kernel.

Ah, OK. thanks.

lurch, 2023-11-06:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes already says "All Raspberry Pi computers report `BCM2835`, even those with BCM2836, BCM2837 and BCM2711 processors. You should not use this string to detect the processor."