raspberrypi/bookworm-feedback #130: FBTFT issue on Bookworm lite only

makermelissa, 2023-10-31:
I am able to set up a TFT with an ILI9331 display that correctly shows the command line. However, once the driver is added to `/boot/config.txt`, removing it does not make the display stop working, so SPI is being used. Is there some new place that bookworm stores drivers that are being loaded?

pelwell, 2023-10-31:
`config.txt` has moved in Bookworm - it now lives in `/boot/firmware`, but with a symbolic link from `/boot/config.txt`. I can imagine situations where an edit operation converts the symbolic link into a real file, at which point its contents are silently ignored.

Otherwise, no - if it's not in the correct config.txt and it's not being loaded due to a HAT EEPROM then you shouldn't be seeing the driver being loaded. `sudo vclog -m` might give you some clues as to what is happening.

6by9, 2023-10-31:
Minor note that we really want to be shifting to the TinyDRM drivers (drivers/gpu/drm/tiny) for these SPI displays, not the fbtft drivers (/drivers/staging/fbtt).
Using DRM drivers should mean that they integrate directly with Wayfire and allow you to extend the desktop to them happily.

Converting the old fbtft overlay to TinyDRM is in progress - see https://github.com/6by9/linux/tree/rpi-6.1.y-tinydrm/arch/arm/boot/dts/overlays, although several of the standalone overlays already have a `drm` override to switch to the DRM drivers (see piscreen, pitft28-resistive, pitft35-resistive, and watterott-display. adafruit-st7735r is TinyDRM only).
(There are various forum threads between aBUGSworstnightmare and myself looking at ILI9341, ST7789, and ST7735 displays).

makermelissa, 2023-10-31:
Thank you. Both bits of info are very helpful.

makermelissa, 2023-10-31:
> I can imagine situations where an edit operation converts the symbolic link into a real file, at which point its contents are silently ignored.

Thanks, I think this is exactly what happened.

> Converting the old fbtft overlay to TinyDRM is in progress - see https://github.com/6by9/linux/tree/rpi-6.1.y-tinydrm/arch/arm/boot/dts/overlays, although several of the standalone overlays already [...]

Thank you. Do you know if converting to tinydrm allow them to work with the desktop environment?

6by9, 2023-10-31:
Minor caveat that some of the displays haven't got an appropriate entry in Mesa and therefore currently fail. I'll be creating a PR for that in due course.

makermelissa, 2024-02-07:
Going to close this as it is no longer an issue.

closed 2024-02-07 by makermelissa