raspberrypi/bookworm-feedback #132: Big mercurial clone gives Bus Error

Spivoxity, 2023-11-01:
When cloning a large repository using mercurial over ssh, the result is a Bus Error.  I'm using 32-bit RPi OS as the client, Debian on amd64 as the host, with identical mercurial versions on both ends.
````
$ hg clone ssh://myserver.net//home/hg/largerepo
destination directory: largerepo
requesting all changes
adding changesets
Bus Error
````
The issue disappears if the --stream flag is used to disable the compact exchange format -- so that's a workaround.  This issue also does not occur with Debian bookworm on amd64 as the client, so it seems to be RPi-specific.  And (sadly for reproducing it) the problem doesn't seem to occur with a small, freshly-made repo, only older and larger ones.

I will try 64-bit RPi OS as the client when I get the chance, and report back.

Spivoxity, 2023-11-04:
I checked with a fresh installation of RPi OS Lite bookworm on RPi 3 and RPi 4, both 32 and 64 bit.  There's no problem with either OS in RPi 3, nor on RPi 4 with 64 bits.  But the 32-bit OS on RPi 4 (which I believe is a 32-bit userland with 64-bit kernel) does exhibit the problem as described.

Spivoxity, 2023-11-04:
Try, for example,
````
$ hg clone https://spivey.oriel.ox.ac.uk/hg/obc-3
````
That's a largish repository that should remain available for the time being.  Access over https gives the bus error just like ssh.

ghollingworth, 2023-11-10:
This sounds like a problem with mercurial when compiled as 32bit userland with a 64-bit kernel, you could try on the Pi 3 with the 64bit kernel, to see if that is the case, but if so it sounds like it needs to go to the devs there.

This doesn't look like an RPiOS issue, so should go upstream...

closed 2023-11-10 by ghollingworth

Spivoxity, 2023-11-10:
Yes, it looks very much like a re-emergence of this problem: https://bz.mercurial-scm.org/show_bug.cgi?id=6715

Spivoxity, 2024-02-20:
A workaround is to disable zstd compression and fall back on zlib.  This will work even when communicating with a server that supports zstd.  Details are here: https://spivey.oriel.ox.ac.uk/corner/Bus_Error_from_Mercurial_on_Raspberry_Pi