raspberrypi/bookworm-feedback #133: Mimo Monitors Vue UM-1080 doesn't display the bookworm desktop

BlackRockCity, 2023-11-02:
I believe bullseye displayed the desktop without issues or additional drivers on this 10.1" touchscreen monitor. I believe the Mimo uses displaylink usb drivers? The new bookworm (64bit or 32bit) release boots into a black screen .

6by9, 2023-11-02:
Can you SSH in and confirm the driver modules loaded via `lsmod`? I suspect it'll be `udl`. If `udlfb` then it is using the old drivers.

If `udl`, then I think it's just a missing entry in Mesa. `ublfb` isn't going to be compatible with Wayfire as it isn't a DRM device.

6by9, 2023-11-07:
I've picked up a UM1080CP-B. 
By default the kernel tries loading both udl and udlfb drivers for the same device - that doesn't help. Create a file `/etc/modprobe.d/blacklist-udlfb.conf` with the contents `blacklist udbfb` and you get only the one come up.

As suspected, udl is missing an entry in Mesa telling it how to handle the device, and that causes Wayfire to abort.
I've got a number of SPI displays that have mainline DRM drivers which are also missing these Mesa entries, so I'm trying to wrap up a big batch of updates adding all those entries.
I'm recompiling Mesa now to see if that one addition makes it all happy.

6by9, 2023-11-07:
OK it does work if Mesa has been told about the device, but it looks to have probably a caching issue as we get mouse pointer trails and corrupted rendering of the desktop (at least on Pi5).
Slightly weirdly, video playback through VLC is perfect, possibly because it is telling Wayfire to redraw the whole frame each update, rather than caching part of the display.

Switching back to X11 is also working, although rather jerkily - that's partly expected as we haven't got a compositor running, and the kernel is having to work fairly hard on the DisplayLink protocol.

kmscube is also giving the same image corruption, so almost certainly it is a lack of cache invalidation before reading what the V3D hardware has generated, or it's not waiting on fences properly and the frame hasn't been finished when it reads it. I'm not sure if that quite works as Wayfire's cursor will be software rendered as there is no cursor plane with udl, so that shouldn't give issues.

BlackRockCity, 2023-11-08:
All I can contribute is that I tried to revert back by loading the legacy image in Raspberry Pi Imager... that didnt load the Mimo screen either. So I ran lsmod and if my feeble brain understands what I saw, it looked like both drivers were loading there as well? Therefore, that "legacy" image has also been recompiled with this unworkable driver situation fairly recently? I know I've seen the Pi Desktop come up on this Mimo monitor as recently as July 2023, without any configuration needed.

6by9, 2023-11-08:
`FB_DRM` was enabled in Nov 2015 on 4.1 via https://github.com/raspberrypi/linux/commit/843d924171c9dd2f85c13800d97485af09b9ba7e

`DRM_UDL` was enabled in 2016 on 4.4 via https://github.com/raspberrypi/linux/issues/1422.

You've had a conflict since 2016, and it's just been a race condition as to which driver was actually active as both udl and udlfb have the same alias of `usb:v17E9p*d*dc*dsc*dp*icFFisc00ip00in*`.

My vote is to kill off `FB_DRM`. DRM emulates a framebuffer anyway, so it shouldn't reduce functionality.

6by9, 2023-11-08:
`FB_UDL` is now removed from the standard kernel via https://github.com/raspberrypi/linux/pull/5702
That gets X11 working again.

I've pushed my mesa patches to https://gitlab.freedesktop.org/6by9/mesa/-/tree/main-kmsro?ref_type=heads, but just want to check something out before creating the merge request. That will get Wayfire working, although with the observed image corruption.

It looks like UDL doesn't support vblank synchronisation, as both kmscube and kmstest report a refresh rate of around 180Hz, even though the display mode is saying 59.99Hz. kmstest is also showing tearing, which is likely down to a real-world mismatch between the permitted rate and the actual rate of the display.

6by9, 2024-07-23:
As far as I'm concerned it's complete.
Mesa changes were merged upstream, and the kernel mods are merged to remove the conflict over which driver would get loaded. I can't say that I've tested it that recently though.

lurch, 2024-07-23:
@BlackRockCity Would you like to test this again with the latest Bookworm OS image after applying all the latest apt updates?

BlackRockCity, 2024-08-11:
Sorry for the delay, when I located a new memory card in a box of junk, I was able to perform a test of the latest Raspberry Pi OS Debian Bookworm 64bit Desktop kernel 6.6.31 5-29-24 release on Raspberry Pi 4 hardware. The display came right up and worked as expected. I updated with `sudo apt update sudo apt upgrade ` then rebooted. Everything continued to work as expected. 

The only issues were numerous pixel artifacts which disappeared when the mouse pointer moved over them. The trail of artifacts was especially evident when moving the mouse over dark mode windows with grey backgrounds.

Also a window header didn't render correctly, the background was offset from the window about 15 pixels to the left. I doubt this has to do with the Displaylink USB driver.. more likely a Wayland compositing bug.

6by9, 2024-08-12:
I'd noted the mouse trail and some corruption in https://github.com/raspberrypi/bookworm-feedback/issues/133#issuecomment-1799296833

IIRC The DisplayLink driver relies heavily on damage tracking so that it only has to update the bits that change.
You also have some fun and games with whether buffers are cached or not, and not all clients perform the correct synchronisation calls.

There are a few other displays that render through almost identical pathways and which are fine (eg DSI and DPI displays on Pi5), so I'm not too concerned that it's a generic bug. The hardware for those directly reads SDRAM for the images though, rather than going through the CPU.

I'd need to retest SPI displays as they also use damage tracking and the CPU to packetise the image. They use a shadow buffer within DRM, whilst UDL doesn't. That would be the first place I'd investigate, but it's not something I'll be looking at.

Closing as it is now working, even if there are minor glitches.

closed 2024-08-12 by 6by9