raspberrypi/bookworm-feedback #135: Chromium with Wayland crashes when browsing boomarks in the bookmarks toolbar

gpi99, 2023-11-06:
Chromium with Wayland crashes when browsing bookmarks in the bookmarks toolbar.  There's a lot of bookmarks spread over the bookmarks toolbar and 10 sub-folders.  The crash can happen quickly within a few seconds of browsing between the folders, or it may take a bit longer browsing, but usually within a half a minute.  When it crashes the Chromium window just disappears completely.  It has to be relaunched from the menu to start again.

The Bookmarks were exported from Chromium on Bullseye to a ´bookmarks_date.html´ file and this was imported into Chromium on Bookworm.  Problem present on freshly imaged bookworm, and also with all updates applied today.  No problem observed with the same Chromium with the bookmarks on X11.

ghollingworth, 2023-11-10:
Cannot reproduce, can you provide bookmarks that reproduce the issue?

closed 2023-11-10 by ghollingworth

gpi99, 2023-11-10:
Just away from the Pi currently, can grab the file next week.