raspberrypi/bookworm-feedback #136: btrfs-convert doesn't work

solskogen, 2023-11-07:
This is special for the Pi5.
btrfs-convert cannot be used because of the page size.
` btrfs-convert /dev/mmcblk0
btrfs-convert from btrfs-progs v6.2

WARNING: blocksize 4096 is not equal to the page size 16384, converted filesystem won't mount on this system`

Normal mkfs.btrfs works.

pelwell, 2023-11-07:
Yes, the same is true of F2FS. If btrfs is important to you, put `kernel=kernel8.img` in config.txt to use the Pi 4 kernel.

solskogen, 2023-11-07:
Yeah, it was more like a FYI rather than a bug. I noticed this when I got my Pi5 yesterday and my old sdcard didn't boot.

closed 2023-11-10 by ghollingworth

leifliddy, 2023-11-29:
There's a discussion on this topic here:
https://social.treehouse.systems/@marcan/111493984306764821