raspberrypi/bookworm-feedback #137: Enabling Read Only File System changes all connected filesystems to read only

wrt54gl, 2023-11-07:
I have been using raspberry pi's with the Overlay read-only file system for some time and have been very happy with it.
In my standard builds I have created another partition on the sd card for any data that I need RW access to then I have just used raspi-config to enable the Overlay File System. 
This has worked well up until Bookworm. In Bookworm, when the read-only file system is enabled it changes all the mounted partitions to read only (except /boot). 
Is there a way to exclude certain partitions or limit it to only the root partition? I cannot seem to find a way to do this.

wrt54gl, 2023-11-07:
Forgot to mention. This using Raspberry pi OS lite 64 bit.

wrt54gl, 2023-11-08:
After much hacking and googling on this:
raspi-config adds `overlayroot=tmpfs` in cmdline.txt to switch the overlayfs on. 
If you change that to `overlayroot=tmpfs:recurse=0` then the overlay file system will behave like before.
You can see all the options in /etc/overlayroot.conf

ghollingworth, 2023-11-08:
Had a chat with Serge about this, since we're now using the standard overlayroot package to do this, we prefer to go with their default options (they've probably thought about the options more than we have!)

You can find more information in /etc/overlayroot.conf of other options available.  We could add an option in raspi-config to select the different behaviour, but it feels difficult to explain the behaviour for most of our users here.  Of course, maybe this all belongs in an "industrial" whitepaper somewhere instead...

closed 2023-11-08 by ghollingworth

lurch, 2023-11-08:
ping @aallan in case he wants to mention this in the [documentation](https://www.raspberrypi.com/documentation/).

JinShil, 2023-11-09:
I think there is still a problem here.  According to the documentation, you should be able to configure the overlay by editing /etc/overlayroot.conf, or better, /etc/overlayroot.local.conf.  So, I think rapi-config should do the following:

1.  Create /etc/overlayroot.local.conf with the line `overlayroot="tmpfs"` if it doesn't exist.
2. If the file does exist, prepend `tmpfs` to the `overlayroot` value.

I don't think raspi-config should modify /boot/cmdline.txt at all.

I tested this myself and it seems to work.

JinShil, 2023-11-09:
Ah, but how does one disable the overlay if one can't write to /etc/overlayroot.local.conf after the overlay is enabled?

Then explicitly adding `overlayroot=""` to /boot/cmdline.txt seems to work, but I hope there is a better way.

XECDesign, 2023-11-09:
> I don't think raspi-config should modify /boot/cmdline.txt at all.

Why not?

wrt54gl, 2023-11-09:
Confusing behavior to say the least. It seems that it needs both /etc/overlayroot.conf and cmdline.txt settings to work currently. 
I thought I had everything set up and then it quit working. I completely re-installed but could not get it to work until I set both places... I would support not using cmdline.txt because there is not much documentation online for that.

XECDesign, 2023-11-09:
Hmm, in my testing, I didn't come across any cases where things didn't "just work" as advertised.

If you can give a set of instructions that get overlayroot to work in a way that doesn't match what's documented in the conf file comments,  I can take a closer look.

wrt54gl, 2023-11-09:
I ran out of time on this project. Seems to need the setting in both places if `overlayroot=tmpfs:recurse=0` is used. Maybe someone else can verify. Otherwise it will be a while before I can verify it.

wrt54gl, 2023-11-09:
> Okay, just did some more testing and having it in either cmdline.txt or overlayroot.conf works.
> 
> cmdline.txt overrides whatever you set in the conf file.
> 
> So, you could have `overlayroot=tmpfs:recurse=0` in the conffile and remove the parameter from cmdline.txt altogether.
> 
> Then if you want to disable it, you can add `overlayroot=disabled` in cmdline.txt.

Makes sense. Thanks for looking into that.

JinShil, 2023-11-09:
> Okay, just did some more testing and having it in either cmdline.txt or overlayroot.conf works.
>
>cmdline.txt overrides whatever you set in the conf file.

Yes, that is my understanding too.

>So, you could have overlayroot=tmpfs:recurse=0 in the conffile and remove the parameter from cmdline.txt altogether.
>
>Then if you want to disable it, you can add overlayroot=disabled in cmdline.txt.

Yes, so to support both the current behavior, and the needs of those who need a read/write partition, please modify raspi-config to enable the overlay by prepending `tempfs` to the user's `overlayroot=` value in /etc/overlayroot.local.conf and then disable it by adding `overlayroot="disabled"` or "overlayroot=""` to /boot/cmdline.txt.

For example, the user might have`overlayroot=":recurse=0"` or other parameters in their /etc/overlayroot.local.conf  file, so raspi-config just needs to prepend `tempfs` to that so it becomes `overlayroot="tempfs:recurse=0"`, enabling the overlay while retaining the user's configuration.

lurch, 2023-11-10:
> The goal is to make raspi-config compatible with the user's configuration in /etc/overlayroot.local.conf somehow.

I think the expectation might be that if you're manually editing `/etc/overlayroot.local.conf`, then you're an advanced user who knows what they're doing, and so you wouldn't **also** be using `raspi-config` to enable the read-only overlay? :thinking:

JinShil, 2023-11-10:
> I think the expectation might be that if you're manually editing /etc/overlayroot.local.conf, then you're an advanced user who knows what they're doing, and so you wouldn't also be using raspi-config to enable the read-only overlay? 

No that's not true.  I create devices for users further down the supply chain to be used as a component in a larger system.  I may be an advanced user, but person I create the device for may or may not be.  Consider compute module-based devices.

The device I create has a boot partition, a read-write partition, and a root partition.  The user further downstream from me in the supply chain may or may not want a read-only file system overlay enabled.  Most likely, after installing their own software they will, but won't know anything about /etc/overlayroot.local.conf or what it should contain and why.  They just know that after enabling the readonly file system overlay using raspi-config or the desktop configuration GUI, the /root partition should be read-only, but the read-write partition should not be.

If raspi-config is not modified to support this, then we need to instruct downstream users to not use raspi-config, and instead use some other method.  But, even more confusingly, to use raspi-config for other things.  They will already be familiar with raspi-config from previous generations and will just expect it to work as it did before, despite our instructions.  This will require us to allocate resources to handle support calls and downstream users will be turned off and annoyed by the whole experience.

wrt54gl, 2023-11-10:
> > I think the expectation might be that if you're manually editing /etc/overlayroot.local.conf, then you're an advanced user who knows what they're doing, and so you wouldn't also be using raspi-config to enable the read-only overlay?
> 
> No that's not true. I create devices for users further down the supply chain to be used as a component in a larger system. I may be an "advanced" user (relatively speaking), but person I create the device for may or may not be. Consider compute module-based devices.
> 
> The device I create has a boot partition, a read-write partition, and a root partition. The user further downstream from me in the supply chain may or may not want a read-only file system overlay enabled. Most likely, after installing their own software they will, but won't know anything about /etc/overlayroot.local.conf or what it should contain and why. They just know that after enabling the readonly file system overlay using raspi-config or the desktop configuration GUI, the /root partition should be read-only, but the read-write partition should not be.
> 
> If raspi-config is not modified to support this, then we need to instruct downstream users to not use raspi-config, and instead use some other method. But, even more confusingly, to use raspi-config for other things. They will already be familiar with raspi-config from previous generations and will just expect it to work as it did before, despite our instructions. This will require us to allocate resources to handle support calls and downstream users will be turned off and annoyed by the whole experience.

Pretty much exactly how I felt!!!

JinShil, 2023-11-27:
FYI:  I've created a pull request at https://github.com/RPi-Distro/raspi-config/pull/225 to propose the aforementioned changes to raspi-config.

haumacher, 2024-10-02:
Why should I want to make an attached HDD or SSD also read-only - only because I want to prevent my SD card from failing every six months? This is a really funny default for me - luckily, I found this issue then...