raspberrypi/bookworm-feedback #142: The 3.3V IO output on the Pi5 + Bookworm is abnormal.

Tiger0215, 2023-11-14:
When an external device with a capacitance exceeding 540uF is connected to the 3.3V IO of the Pi5, the system fails to start. The power indicator light remains red, and the measured voltage on the 3.3V pin is 0V.

pelwell, 2023-11-14:
When you say "3.3V IO", do you mean pin 1 of the 40-pin header, one of the other (GPIO) pins on the header, or something else?

Tiger0215, 2023-11-14:
pin 1 of the 40-pin header

pelwell, 2023-11-14:
Just because you got away with it on Pi 4 doesn't mean it's a good idea. What are you trying to achieve?

Tiger0215, 2023-11-14:
Due to the presence of many external devices working normally on Raspberry Pi 4, an issue with capacitor detection prevents the proper startup on Pi 5. Under the red light state, it requires pressing the power button on Pi 5 again to start the system normally. This poses some challenges for our industrial automation use.

We hope that Pi 5, as an upgraded version of Pi 4, can maximize compatibility with devices supported by Pi 4. We acknowledge your statement that Pi 5 enters a protective state for security reasons. At the same time, we also hope for a method to manually enable the maximum allowed capacitance on Pi 5, achieving the desired effect of normal power-on startup. This would greatly benefit our large industrial equipment.

byzantic, 2023-11-14:
I think what was meant, was what is the purpose of such a large capacitance on the 3.3V I/O pin? We all understand that you just want everything to run as it did previously..

As stated, 540 uF is a _very_ large value. The 3.3V I/O supply is intended as an I/O _reference_ voltage, but can run some light loads as well. I can't find any specs, but knowing Dialog/Renesas, it can probably supply more current than the Pi 4.

So 540 uF should not be needed to provide decoupling on a digital supply like that. I can think of a number of reasons why you might have that value:
1. There are multiple boards / systems connected to the supply, each with their own decoupling
2. There are some analog circuits connected to the supply that are sensitive to interference, and the decoupling has been added to reduce sensitivity.

Both of these situations are really systems issues; you might need to look at the system again. By the way, I couldn't find a specification for the available current on the 3.3V pin.

If it's something else you would probably need to be more specific. 

As an aside, this is not really a Bookworm issue, it's a hardware issue, but I'm not sure if there is a specific github project available to track those ..

pelwell, 2023-11-15:
The maximum capacitance on Pi 5's 3V3 pin will not change, but it will be documented in the HAT spec - probably as 100uF.

closed 2023-11-15 by pelwell