raspberrypi/bookworm-feedback #143: Image does not appear via HDMI after boot

opastorello, 2023-11-14:
The desktop is no longer appearing after booting, does anyone have a solution?

popcornmix, 2023-11-14:
It's probably best to ask this question on the [forum](https://forums.raspberrypi.com/viewforum.php?f=28),
as there are common reasons why this fails (e.g. faulty hdmi cable).

If a software bug is confirmed we can reopen this issue.

closed 2023-11-14 by popcornmix

opastorello, 2023-11-15:
> Provavelmente é melhor fazer esta pergunta no [fórum](https://forums.raspberrypi.com/viewforum.php?f=28) , pois há motivos comuns para isso falhar (por exemplo, cabo HDMI com defeito).
> 
> Se um bug de software for confirmado, podemos reabrir este problema.

The problem is not with the HDMI cable, I try to see the boot image but the screen goes black when the boot starts and the Raspian desktop appears.

popcornmix, 2023-11-15:
Make a forum post and we'll try to narrow down where the problem is.