raspberrypi/bookworm-feedback #144: kodi-inputstream-rtmp is not installable on `armhf`

cmitu, 2023-11-14:
The Raspbian repository contains a newer version than the `kodi-inputstream-rtmp` package from both Raspberry Pi repository.

```
# kodi-inputstream-rtmp
apt-cache policy kodi-inputstream-rtmp
kodi-inputstream-rtmp:
  Installed: (none)
  Candidate: 20.3.0+ds1-1
  Version table:
     20.3.0+ds1-1 500
        500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
     20.3.0-2~bookworm 500
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
```

Trying to install `kodi-inputstream-rtmp` results in an error, since it references a non-existent dependency (`kodi-api-inputstream`):

```
shell $ sudo apt install kodi-inputstream-rtmp
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kodi-inputstream-rtmp : Depends: kodi-api-inputstream (>= 3.2.0)
                         Depends: kodi-api-inputstream (< 3.3.0)
```

aandroide, 2023-12-09:
Hi, I have the same problem too. iptvsimpleclient is installed but does not install the rtmp and ffmpegdirect dependencies. same errors as above.

aandroide, 2023-12-18:
News?

cmitu, 2023-12-19:
@aandroide you can always install the version you want by specifying it after the package name, e.g.
``` sh
sudo apt install kodi-api-inputstream=20.3.0-2~bookworm
```

lurch, 2023-12-19:
On 32-bit Bookworm I get:
```
pi@bookworm32:~ $ apt policy kodi-inputstream-rtmp
kodi-inputstream-rtmp:
  Installed: (none)
  Candidate: 20.3.0+ds1-1
  Version table:
     20.3.0+ds1-1 500
        500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
     20.3.0-2~bookworm 500
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
pi@bookworm32:~ $ apt show -a kodi-inputstream-rtmp
Package: kodi-inputstream-rtmp
Version: 20.3.0+ds1-1
Priority: optional
Section: video
Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Installed-Size: 322 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.5), librtmp1 (>= 2.3), libstdc++6 (>= 6), kodi-api-inputstream (>= 3.2.0), kodi-api-inputstream (<< 3.3.0)
Homepage: https://kodi.tv/
Download-Size: 77.5 kB
APT-Sources: http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
Description: Kodi input stream addon for RTMP
 This package is the RTMP Inputstream addon for Kodi.
 .
 The Real Time Messaging Protocol (RTMP) is a proprietary network protocol
 developed by Adobe Inc. to transmit audio, video and other data over the
 Internet from a media server to a flash player.

Package: kodi-inputstream-rtmp
Version: 20.3.0-2~bookworm
Priority: extra
Section: libs
Maintainer: Arne Morten Kvarving <spiff@xbmc.org>
Installed-Size: 299 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.5), librtmp1 (>= 2.3), libstdc++6 (>= 6)
Download-Size: 75.1 kB
APT-Sources: http://archive.raspberrypi.com/debian bookworm/main armhf Packages
Description: Rtmp inputstream addon for Kodi
 This is the RTMP inputstream addon for Kodi

pi@bookworm32:~ $ apt show -a kodi-api-inputstream
Package: kodi-api-inputstream
State: not a real package (virtual)
N: Can't select versions from package 'kodi-api-inputstream' as it is purely virtual
N: No packages found
```

and on 64-bit Bookworm I get:
```
pi@bookworm64:~ $ apt policy kodi-inputstream-rtmp
kodi-inputstream-rtmp:
  Installed: (none)
  Candidate: 20.3.0+ds1-1
  Version table:
     20.3.0+ds1-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
     20.3.0-2~bookworm 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
pi@bookworm64:~ $ apt show -a kodi-inputstream-rtmp
Package: kodi-inputstream-rtmp
Version: 20.3.0+ds1-1
Priority: optional
Section: video
Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Installed-Size: 322 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.0), librtmp1 (>= 2.3), libstdc++6 (>= 6), kodi-api-inputstream (>= 3.2.0), kodi-api-inputstream (<< 3.3.0)
Homepage: https://kodi.tv/
Download-Size: 80.1 kB
APT-Sources: http://deb.debian.org/debian bookworm/main arm64 Packages
Description: Kodi input stream addon for RTMP
 This package is the RTMP Inputstream addon for Kodi.
 .
 The Real Time Messaging Protocol (RTMP) is a proprietary network protocol
 developed by Adobe Inc. to transmit audio, video and other data over the
 Internet from a media server to a flash player.

Package: kodi-inputstream-rtmp
Version: 20.3.0-2~bookworm
Priority: extra
Section: libs
Maintainer: Arne Morten Kvarving <spiff@xbmc.org>
Installed-Size: 321 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), librtmp1 (>= 2.3), libstdc++6 (>= 6)
Download-Size: 77.5 kB
APT-Sources: http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
Description: Rtmp inputstream addon for Kodi
 This is the RTMP inputstream addon for Kodi

pi@bookworm64:~ $ apt show -a kodi-api-inputstream
Package: kodi-api-inputstream
State: not a real package (virtual)
N: Can't select versions from package 'kodi-api-inputstream' as it is purely virtual
N: No packages found
```

Looks like this might be a packaging-problem for @XECDesign to address?

aandroide, 2023-12-19:
> @aandroide you can always install the version you want by specifying it after the package name, e.g.
> 
> ```shell
> sudo apt install kodi-api-inputstream=20.3.0-2~bookworm
> ```
sudo apt install kodi-api-inputstream=20.3.0-2~bookworm
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package kodi-api-inputstream is a virtual package provided by:
  kodi 2:20.1+dfsg-1 (= 3.2.0) [Not candidate version]

E: Version '20.3.0-2~bookworm' for 'kodi-api-inputstream' was not found

JuenTingShie, 2024-01-28:
related to `kodi-inputstream-ffmpegdirect`, this package also not found.

I'm using fresh install of [2023-12-05-raspios-bookworm-arm64.img.xz](https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2023-12-06/2023-12-05-raspios-bookworm-arm64.img.xz)

cmitu, 2024-01-28:
As a workaround, you can use an pin file for `apt` to priorize packages from the RPF repos.

Add to `/etc/apt/preferences.d/01-rpf-kodi`:
```
Package: kodi*
Pin: release o=Raspberry Pi Foundation
Pin-Priority: 900
```

JuenTingShie, 2024-01-28:
> As a workaround, you can use `apt` pinning to prioritize packages from the RPF repos.
> 
> Add to `/etc/apt/preferences.d/01-rpf-kodi`:
> 
> ```
> Package: kodi*
> Pin: release o=Raspberry Pi Foundation
> Pin-Priority: 900
> ```

Hi, I have read some documents of APT pinning, but could you please explain simply why to set up pin?

cmitu, 2024-01-28:
> Hi, I have read some documents of APT pinning, but could you please explain simply why to set up pin?

_Why_ is this issue itself - due to version overlap, `kodi` is not directly installable on RaspiOS. See the 1st post of this issue.

spookyfirehorse, 2024-06-09:
> The Raspbian repository contains a newer version than the `kodi-inputstream-rtmp` package from both Raspberry Pi repository.
>
> ```
> # kodi-inputstream-rtmp
> apt-cache policy kodi-inputstream-rtmp [...]

sudo apt install kodi21-inputstream-*  kodi21   installs the latest kodi  but do not make an versions update in kodi there is a bug and it install the normal kodi package

spookyfirehorse, 2024-06-10:
> > sudo apt install kodi21-inputstream-* kodi21 installs the latest kodi ...
> 
> Not sure if this it's related to the issue at hand. The naming of the new packages is on purpose so that they could be installed separately (though not at once) - so `kodi21...` will not upgrade the `kodi..` packages.

Hello

Versions conflickt

sudo apt install kodi21-inputstream-* kodi21

Try it but when you start kodi21 don't do a versions check update it will reinstall kodi's old version but only kodi and not kodi-inputstream

so don't do a versions check update in kodi direct when it's started only via apt

sudo apt install kodi-inputstream-* kodi

or

sudo apt install kodi21-inputstream-* kodi21

 this works try it