raspberrypi/bookworm-feedback #145: OS root/boot archives are not updated

mdevaev, 2023-11-15:
Here:
* https://downloads.raspberrypi.com/raspios_lite_arm64/
* https://downloads.raspberrypi.com/raspios_lite_armhf/
* https://downloads.raspberrypi.com/raspios_full_arm64/
* https://downloads.raspberrypi.com/raspios_full_armhf/

Files `boot.tar.xz` are `root.tar.xz` are not updated and belongs to bullseye. `os.json` is not updated too and points ro previous image release.

lurch, 2023-11-15:
Those files were only used by [NOOBS](https://github.com/raspberrypi/noobs), and NOOBS got deprecated a _long_ time ago, so those files are no longer being updated.

lurch, 2023-11-15:
Duplicate of #78

closed 2023-11-15 by lurch

mdevaev, 2023-11-15:
@lurch is it possible to provide complete tar archive for whole rootfs besides img files? This would be handy for creating docker images. To convert an img to a root archive, we need to perform losetup and many other actions.