raspberrypi/bookworm-feedback #146: Pi4B running Bookworm closing processes

Gordon999, 2023-11-16:
My Pi4B running 64bit Bookworm will close running processes if the hdmi monitor is switched off and on, asking me to sign in again.

I first noticed this when using python and pygame where if running through IDLE it would close python and ask me to sign in when the hdmi monitor was switched off and on. Testing similar through Thonny would corrupt the pygame window graphics, but l have also seen this stop python and ask me to sign in again. 

If l don't sign in again and switch the monitor off and on again this results ina black screen with mouse cursor, which only a reboot appears to resolve.

JamesH65, 2023-11-16:
What is the make of the monitor? I tried tried this with a Dell 4k monitor, and did not see the same issue. I tried the monitor on/off and also pulled the HDMI lead out and reinserted.

I suspect something to do with the HDMI hotplug and EDID reading is crashing, causing Wayfire to crash. It restarts and puts you back to the login prompt. Some child processes would be closed on the way I suspect.

popcornmix, 2023-11-16:
@Gordon999 is it only pygame that does this?
I just tried with chromium and neverball running and they are still running after an on/off and after a hdmi cable unplug/replug.

popcornmix, 2023-11-16:
Also tried with `sudo apt install code-the-classics` then launched boing.
`/usr/share/code-the-classics/boing/boing.py` does use pygame.

Yes - that does behave badly.
Looked like boing was running shortly after reinsterting hdmi but I was kicked back to login screen soon after.

Another power cycle and I have a blank screen.

`jouralctl -e` ends with:
```
Nov 16 14:16:05 pi5 systemd[925]: xdg-desktop-portal-wlr.service: start operation timed out. Terminating.
Nov 16 14:16:05 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Failed with result 'timeout'.
Nov 16 14:16:05 pi5 systemd[925]: Failed to start xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Nov 16 14:16:05 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 3.
Nov 16 14:16:05 pi5 systemd[925]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Nov 16 14:16:05 pi5 systemd[925]: Starting xdg-desktop-portal-wlr.service - Portal service (wlroots implementation)...
Nov 16 14:16:05 pi5 xdg-desktop-portal-wlr[2428]: 2023/11/16 14:16:05 [ERROR] - wayland: failed to connect to display
Nov 16 14:16:05 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 14:16:05 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
...
Nov 16 14:16:06 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 8.
Nov 16 14:16:06 pi5 systemd[925]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Nov 16 14:16:06 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Start request repeated too quickly.
Nov 16 14:16:06 pi5 systemd[925]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
Nov 16 14:16:06 pi5 systemd[925]: Failed to start xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
```

Gordon999, 2023-11-16:
> What is the make of the monitor? I tried tried this with a Dell 4k monitor, and did not see the same issue. I tried the monitor on/off and also pulled the HDMI lead out and reinserted.
>
> I suspect [...]

LG 23" HD monitor

Gordon999, 2023-11-17:
I guess this issue also relates to an issue l had with pygame and a 4" lcd...

https://forums.raspberrypi.com/viewtopic.php?t=359402

RobertAllan-Alien-Dad, 2023-12-19:
This is causing me an issue too I believe.  I think there are two issues.  One may be working as needed for good security practice, but I hope not. Problem 1: restarting the xdg-desktop-portal-wlr causes a login prompt. 2. there appears to be a scheduled restart of xdg-desktop-portal-wlr in rpios 64b for rpi4, which is maybe to help with a reported video memory leak with xdg-desktop-portal-*  I found it in the sysctl messages via journalctl

lurch, 2023-12-19:
I've just tried testing this on the latest fully-up-to-date Bookworm (both 32-bit and 64-bit, on both a Pi5 and a Pi4) and with turning the HDMI monitor off & on and/or unplugging and replugging the HDMI cable, I'm unable to get it to kick me back to the login prompt. Also tried doing the same with Boing running, and that also still kept running after unplugging and replugging the HDMI cable. Are you still able to recreate this @popcornmix , or has "something" been updated in the meantime which makes things more reliable now?

One thing I **did** notice was that _sometimes_ after unplugging and replugging the HDMI cable, the taskbar-panel (`wf-panel-pi`) dies and gets restarted - although unfortunately this seems to happen randomly and isn't 100% reproducible.

lurch, 2023-12-19:
I have just spotted one other "quirk" related to switching monitors off and on again though:

Raspberry Pi 4, running fully up-to-date 64-bit version of Bookworm, HDMI monitors plugged into both the HDMI0 ("left monitor") and HDMI1 ("right monitor") ports.
At initial power-on, the taskbar is displayed on the left monitor. If I have e.g. a Terminal window open on the left monitor and a File Manager window open on the right monitor, if I power-off the right monitor, the left monitor remains unchanged. If I then power **on** the right monitor, the File Manager suddenly jumps over to the left monitor!
If I then move the File Manager back to the right monitor, and power off the left monitor, then the File Manager is still visible on the right monitor (and I can continue to interact with it with the mouse). But as soon as I power the left monitor back on again, both the Terminal window **and** the taskbar suddenly jump over to the right monitor!

popcornmix, 2023-12-19:
> Are you still able to recreate this @popcornmix , or has "something" been updated in the meantime which makes things more reliable now?

I've just tried and I found launching Boing!, unplugging and replugging hdmi cable resulted in:
- Boing! disappearing on Pi4
- Boing! still running on Pi5

I then did an apt full-upgrade on pi4 which was a little out of date (apt showed these as having updates)
[code]
  base-files chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra cups cups-client cups-common cups-core-drivers cups-daemon cups-ipp-utils cups-ppdc cups-server-common devscripts dillo distro-info-data exfatprogs
  ghostscript gir1.2-gtk-3.0 gir1.2-handy-1 gtk-update-icon-cache libcups2 libcupsimage2 libde265-0 libdistro-info-perl libgnutls30 libgs-common libgs10 libgs10-common libgtk-3-0 libgtk-3-common libhandy-1-0 libjavascriptcoregtk-4.1-0
  libpam-systemd libperl5.36 libqpdf29 libsystemd-dev libsystemd-shared libsystemd0 libudev-dev libudev1 libvlc-bin libvlc5 libvlccore9 libwebkit2gtk-4.1-0 perl perl-base perl-modules-5.36 pixflat-icons pixflat-theme
  raspberrypi-sys-mods raspi-config rpi-eeprom systemd systemd-sysv systemd-timesyncd tzdata udev vlc vlc-bin vlc-data vlc-l10n vlc-plugin-access-extra vlc-plugin-base vlc-plugin-notify vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2
  vlc-plugin-video-output vlc-plugin-video-splitter vlc-plugin-visualization xserver-common xserver-xorg-core
[/code]
and after a reboot Boing! is still running after a hdmi unplug/replug.

So I think I agree, this seems to be fixed in an update.

@Gordon999 can you still reproduce on an updated bookworm system?

lurch, 2023-12-19:
>  there appears to be a scheduled restart of xdg-desktop-portal-wlr in rpios 64b for rpi4

@RobertAllan-Alien-Dad I've just had 64-bit Bookworm running on a Pi 4B (running a Terminal and File Manager window) for over two hours with no problems, so I suspect this "scheduled restart" might be a red herring?

RobertAllan-Alien-Dad, 2023-12-21:
I agree.  Now I am trying to figure out how to add the option to turn the xfg-desjtop-portal-wlr loglevel on to "trace" so I can figure out what actually brought it down. Maybe an unusual error in an image or video file?

hughjenjin, 2023-12-24:
Just updated to latest OS version and have the same issue. Have done an update/upgrade and only installed Thunderbird but nothing else.
 Not every time but quite frequently after turning the monitor off and on again all open applications have closed. The Pi has not rebooted but i don't know how to see if the desktop has restarted?
 If the screen blanking kicks in it also closes all open applications but again not every time but frequently.
 The monitor is a 27" Acer connected with a known good lead and the issue happens with either HDMI port on the Pi.

 Pi 4 8GB with official PSU
Have tried running from SD card and external SSD
Same setup i have been using for years and the issue only started immediately after updating to Bookworm.

 Anybody able to offer any help? If people are already on the case i will obviously be patient and hope for as quick fix.

PS. I was going to try to use a HDMI-DVI adapter to see if that made it better but unfortunately this monitor doesn't have DVI. Has anybody else tried that option?

popcornmix, 2023-12-24:
I would expect adding:
`vc4.force_hotplug=1` for HDMI-A-1
`vc4.force_hotplug=2` for HDMI-A-2
`vc4.force_hotplug=3` for HDMI-A-1 and HDMI-A-2

to cmdline.txt (on the existing line) and rebooting will offer a workaround to this issue.

hughjenjin, 2023-12-28:
Thanks for that popcornmix, It's been 48 hours and not a single issue yet where i was getting an issue several times a day.
 I can recommend this for anyone else with the same problem.

Gordon999, 2024-01-01:
> > Are you still able to recreate this @popcornmix , or has "something" been updated in the meantime which makes things more reliable now?
> 
> I've just tried and I found launching Boing!, unplugging and replugging hdmi cable resulted in:
> 
> * Boing! disappearing on Pi4
> * Boing! still running on Pi5
> 
> I then did an apt full-upgrade on pi4 which was a little out of date (apt showed these as having updates)
> 
> ```
>   base-files chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra cups cups-client cups-common cups-core-drivers cups-daemon cups-ipp-utils cups-ppdc cups-server-common devscripts dillo distro-info-data exfatprogs
>   ghostscript gir1.2-gtk-3.0 gir1.2-handy-1 gtk-update-icon-cache libcups2 libcupsimage2 libde265-0 libdistro-info-perl libgnutls30 libgs-common libgs10 libgs10-common libgtk-3-0 libgtk-3-common libhandy-1-0 libjavascriptcoregtk-4.1-0
>   libpam-systemd libperl5.36 libqpdf29 libsystemd-dev libsystemd-shared libsystemd0 libudev-dev libudev1 libvlc-bin libvlc5 libvlccore9 libwebkit2gtk-4.1-0 perl perl-base perl-modules-5.36 pixflat-icons pixflat-theme
>   raspberrypi-sys-mods raspi-config rpi-eeprom systemd systemd-sysv systemd-timesyncd tzdata udev vlc vlc-bin vlc-data vlc-l10n vlc-plugin-access-extra vlc-plugin-base vlc-plugin-notify vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2
>   vlc-plugin-video-output vlc-plugin-video-splitter vlc-plugin-visualization xserver-common xserver-xorg-core
> ```
> 
> and after a reboot Boing! is still running after a hdmi unplug/replug.
> 
> So I think I agree, this seems to be fixed in an update.
> 
> @Gordon999 can you still reproduce on an updated bookworm system?

Yes. Still there when using pygame.