raspberrypi/bookworm-feedback #147: Wayland + Desktop without autologin opens lightdm greeter only on HDMI port 1

josefwegner, 2023-11-18:
If Wayland is enabled and you configure boot to desktop without autologin, lightdm will be opened on physical HDMI port 1 (HDMI1-A-2) regardless if there is a monitor connected or not.

I had my display connected on physical port HDMI 0 (the one near the power port), and I got just a gray screen with a mouse pointer on boot.

After connecting the monitor to the other port (HDMI 1), I see the login prompt after boot.

josefwegner, 2023-11-18:
Please disregard my initial error description. The root cause is not the port but the selected resolution.

If the resolution you selected is not the same as the max resolution supported by the monitor then the login will not be displayed.

I am using a 4k monitor, but changed the resolution to 1920x1080x60Hz for performance reasons. The breaks the login.

It worked once after changing the HDMI ports because the display started up in 3840x2160x30Hz. After changing the resolution back to Full HD, the login prompt again fails to show.

spl237, 2023-12-07:
I've just tested this here - with my 4K monitor set to 1920x1080 in the Screen Configuration tool, and booting to desktop without autologin, I get the greeter displayed correctly without any problems. The greeter is displayed in an identical wayfire environment to the actual desktop - the only real difference is that fewer plugins are loaded - and the setting of screen resolution is done identically for the greeter as for the desktop, so there is no reason for there to be any difference.

If there is a problem here, it is unlikely to be anything to do with the greeter, but more likely to be a generic HDMI issue.

andrum993, 2023-12-07:
> The only way to sign back into the Pi is to power cycle it, which is unfortunate as the official case does not have a power button.

It does on a Pi 5.

I suspect the problems noted above are specific to certain monitors. I had a minor issue with my system under Wayland that was down to the monitor. I suspect that if you changed from Wayland to X11 that would fix the problems you have both experienced. You can switch to X11 using the `raspi-config` utility: head to the `6 Advanced Options` menu, then choose `A6 Wayland`, then select `W1 X11`. Note that you will lose hardware graphics acceleration, but that would at least confirm that the issue is Wayland specific. You can then switch back to Wayland using `raspi-config`.

josefwegner, 2023-12-07:
The issue only happens when using Wayland. Switching to X11 will show the login mask every time. On a side note, display blanking also only works with X11, not Wayland.

I am using a DELL UP2414Q connected via HDMI KVM switch.

tehstu, 2023-12-07:
> It does on a Pi 5.

Suspect I'm missing something, but I do not see a power button on the official Pi 5 case. I digress, it's just a temporary case for me.

> I suspect the problems noted above are specific to certain monitors. I had a minor issue with my system under Wayland that was down to the monitor. I suspect that if you changed from Wayland to X11 [...]

Yes, this is Wayland specific. I switched to X11 and no longer have this issue. I will remain on X11 until RealVNC figure out Wayland support, plus this issue.

For reference, my monitor is a Dell P3421W. Just to clarify: 

**Steps to reproduce:**
- Power on Pi, connected via official HDMI lead
- Desktop loads as expected, showing login screen (autologin is disabled)
- Sign in, sign out
- Turn off monitor
- Turn on monitor

**Observed**:

- Greeter not displayed
- Mouse input works, I can see and move the cursor
- Keyboard input does nothing

tehstu, 2023-12-07:
> I am using a DELL UP2414Q connected via HDMI KVM switch.

That's interesting. My monitor also has a KVM, but I'm not "using" that aspect of it. The issue manifests without me switching input source. I have a spare, non-KVM Dell U2415. I will test with this monitor as soon as I'm able.

lurch, 2023-12-09:
"mouse input works, keyboard input does nothing" also smells suspiciously like a KVM problem? AFAIK there's no reason why switching between Wayland and X11 should have any affect on the functionality of your USB keyboard?

tehstu, 2023-12-09:
> "mouse input works, keyboard input does nothing" also smells suspiciously like a KVM problem? AFAIK there's no reason why switching between Wayland and X11 should have any affect on the functionality of your USB keyboard?

When the greeter does not load, there are no controls rendered for the keyboard to provide any input to. "Keyboard input does nothing" implied I was unable to jog things into working again (CTRL+C, ESC, etc.).

Here is the extent of my testing, which I believe also rules out the KVM:

Keyboard - Cherry MX Board 3.0 S (wired)
Mouse - Logitech M510 (wireless, unified Logitech receiver)
Pi 5 8gb official case
Boot from Sabrent USB3 enclosure (Intel 320 80gb SSD)
Official Pi HDMI cable

**Steps to reproduce**:
1. Enable Wayland, reboot
2. Sign in
3. Sign out
4. Turn off monitor
5. Turn on monitor

**Observed**:

- Greeter does not load, screen is shaded dark grey (I don't think it's my backlight showing black) and a working mouse pointer
- Power button on official Pi 5 case no longer gracefully shuts down system (holding it down will eventually cause power off)

Hardware combinations resulting in observed behavior:

**Dell P3421W** (running native 3440x1440)
Test 1
- Mouse and keyboard connected to monitor (KVM set to HDMI)
- Monitor's USB connected to Pi
- Pi HDMI port 0

Test 2

- Mouse and keyboard connected to Pi
- Monitor's USB connected to Pi
- Pi HDMI port 0

**Dell U2415** (running native 1920x1200)
Test 3

- Mouse and keyboard connected to Pi
- Pi HDMI port 0

Test 4

- Pi HDMI port 1 (just to rule out port 0 acting up

What are next steps here? There are a couple of ASUS monitors in the house I could try, or perhaps TVs of varying resolution. That it's 3 different monitors so far (all Dell, but only two with KVM) suggests it would be helpful to start looking at whatever error logs might point to the issue. Any pointers appreciated.

spl237, 2023-12-10:
Tried this here - my Pi is connected to an Iiyama 4K monitor via a CKLau 4-port KVM. I do not see the problem you describe - I can power the monitor off and back on again at the greeter, and the greeter is still there when it comes back on. I can also boot with the monitor off, turn the monitor on when the greeter has loaded, and it again works fine.

This really does seem to be something specific to your HDMI configuration. It is possible that something in the hotplug sequence generated by your monitor crashes Wayfire; that, I think, would lead to the symptoms you describe, but it would be unusual for it to only happen on the greeter; I would expect it to happen just the same on the desktop.

Our monitors in the office are also Dell 4K, and I do not recall seeing this problem with them, but I cannot say for sure that I have tried these exact steps. But certainly with an Iiyama monitor and a CKLau KVM, the problem does not occur.

tehstu, 2023-12-10:
> Tried this here - my Pi is connected to an Iiyama 4K monitor via a CKLau 4-port KVM. I do not see the problem you describe - I can power the monitor off and back on again at the greeter, and the [...]

Yes, I think the presence of a built-in KVM is a red herring. While on the desktop in Wayland, I actually invoked the KVM by changing input source over to USB-C (work laptop). Upon returning, the desktop displayed correctly and both keyboard and mouse worked as expected.

> This really does seem to be something specific to your HDMI configuration. It is possible that something in the hotplug sequence generated by your monitor crashes Wayfire; that, I think, would lead [...]

I've confirmed it's limited to the greeter screen. While on the desktop, I've turned the monitor on and off multiple times, altering the delay on each occasion, and the desktop always returns. This issue only manifests when on the greeter.

It also manifests if the monitor is not powered on at boot. If I let the machine boot and turn the monitor on after a short while, I get the same dark grey background with working mouse, but no greeter.

Glancing through `journalctl` I noted a few warnings related to Pipewire, but those seem to be present even on a successful boot. On a hunch, I disabled HDMI audio (set to off) but this did not resolve greeter failing to render.

spl237, 2023-12-10:
What you could try doing to see if it helps is to edit /etc/wayfire/greeter.ini and add some of the plugins you will find in /etc/wayfire/defaults.ini. Try copying the plugins= line from defaults.ini and use it in greeter.ini - see if that helps. If it does, it should then be relatively easy to go through and delete plugins until it stops working again to narrow down which plugin makes it work.

The plugins are the only real difference between the greeter and the desktop wayfire environment, so if the greeter has the problem and the desktop doesn't, it may be because a plugin is fixing it. Let us know how you get on!

juangentile, 2024-05-09:
I wonder if someone was able to fix this. In my case I have 2 inputs connected to my screen and when I switched to the other one (having logged off the rpi) and then come back to it I have the problem. The "quick" workaround I found instead of rebooting is to simply restart the lightdm service (`sudo systemctl restart lightdm.service`) via ssh from another machine.

josefwegner, 2024-05-09:
There seems to be a workaround that fixed the issues for me. Please append the following to `/boot/firmware/cmdline.txt
`:
`vc4.force_hotplug=1`

This is for a monitor connected to monitors to HDMI 1 (the one nearest to power).

Please do not ask me what exactly this is doing, as I understand it, it disables disconnect detection of monitors (although the command would indicate the opposite). At least for me, no more crashes of apps, random shutdowns (grey screens), or grey screens on boot up.

popcornmix, 2024-05-09:
@josefwegner 

`vc4.force_hotplug=1` means the value the kernel reads from the hotplug pin of the hdmi is forced to 1, so the kernel cannot detect a monitor disconnect.

Can you confirm what situation setting this fixes? It should only make a difference when the monitor is sometimes signalling it is disconnected. Does this avoid problem when you are powering off and on a monitor (with the Pi desktop running)?

josefwegner, 2024-05-09:
Without this setting, I had the following issues:

- booting up the Pi without switching to the Pi on the KVM would result in a grey screen (100% of the time)
- switching from the Pi to a different computer and back while having maximized apps would crash the panel (100%)
- switching from the Pi and back would kill random non-maximized apps or the complete session (20%)

These issues went away with that setting.

spl237, 2024-05-09:
> Without this setting, I had the following issues:
> 
> * booting up the Pi without switching to the Pi on the KVM would result in a grey screen (100% of the time)
> * switching from the Pi to a different computer and back while having maximized apps would crash the panel (100%)

That sounds to me like a dodgy KVM switch. I use a KVM to switch between Mac, PC and Pi dozens of times every day, and have never seen either of those problems.

josefwegner, 2024-05-09:
I used two different KVMs and three different monitors, all caused the same issue.

and of the KVMs was 250€, so not a top level device, but still expensive enough.

juangentile, 2024-05-09:
Thank you @josefwegner and @spl237 for the quick feedback! I'll give it a try!

popcornmix, 2024-05-09:
@josefwegner can you just confirm that after apt updating and with `vc4.force_hotplug=1` removed that the issue is still present?

There was an issue a month or two back where putting a display into standby (which on some displays removes hotplug) caused a crash, but that has been resolved so I want to confirm you've tested since then.

josefwegner, 2024-05-09:
@popcornmix 
Sorry, I do not understand your question.

I had the listed issue until roughly last week or the week before last week, when I added the flag to cmdline.txt.

So, I had the issues until very recently. Before I mostly used X11, and occasionally switched back to Wayland to check if there is bug fix (spoiler: there wasn’t).

popcornmix, 2024-05-09:
Make sure you are updated (`sudo apt update && sudo apt full-upgrade -y`)
Make sure you are using the default wayland configuration.
Remove `vc4.force_hotplug=1`
Reboot.

Log in. Without running anything else, what happens when you switch away and back with KVM switch?
If there is an issue, does it occur sometimes or always?

popcornmix, 2024-05-09:
> But no issues when I just log in and do not start any programs. But that is the sense for that?

Okay, what is the minimum you can do to show the problem?

e.g. with KVM showing Pi, you boot, you log in (or auto log in), you launch one of the stock applications (e.g. chromium), maximise it, switch KVM away and back to Pi - does that always crash something?

Can you ssh in and capture url from:
```
raspinfo | pastebinit
```
after boot with KVM showing Pi display, and then again with KVM switched away from Pi display.
I want to confirm the suspicion that KVM deasserts hotplug when switched away (which I'm guessing @spl237's may not).

josefwegner, 2024-05-09:
> Okay, what is the minimum you can do to show the problem?

- Boot with KVM showing Pi
- Log in
- Launch Chromium
- Maximize Chromium
- Switch KVM away and back to Pi

This results in crashing the panel to 100% (every freaking time). How do I know that? Because the panels shows tool tips for my internet connection and available updates after switching back.
 
> Can you ssh in and capture url from:
> 
> ```
> raspinfo | pastebinit
> ```
After boot with KVM showing Pi and logging in:
https://paste.debian.net/hidden/42a7c284/
After switching to different machine and ssh into the Pi:
https://paste.debian.net/hidden/1d25e89e/
After switching back to Pi:
https://paste.debian.net/hidden/38b3f89e/

josefwegner, 2024-05-09:
Strange, it seems another workaround is to unplug the HDMI cable from Pi to KVM switch and plug it in again. After that the panel is no longer crashing after switching.

spl237, 2024-05-09:
Have you tried a different cable? Might be that the plug connection or wiring in that one is glitching.

josefwegner, 2024-05-09:
Yes, I tried different cables for both the connection from the Pi to the KVM (both are official Raspberry Pi mini HDMI to HDMI cables) and KVM to monitor (different HDMI to HDMI cables).

popcornmix, 2024-05-13:
>    Boot with KVM showing Pi
>    Log in
>    Launch Chromium
>    Maximize Chromium
>    Switch KVM away and back to Pi
>
> This results in crashing the panel to 100% (every freaking time). How do I know that? Because the panel shows tool tips for my internet connection and available updates after switching back.

I can't reproduce this by unplugging/replugging hdmi cable, or just disconnecting hotplug detect line.

@josefwegner does uplugging/replugging the hdmi cable cause the issue for you, or does it recover just fine?

josefwegner, 2024-05-13:
> @josefwegner does uplugging/replugging the hdmi cable cause the issue for you, or does it recover just fine?

No, it is the other way around. I have the issues without unplugging/replugging.

Unplugging/replugging seems to temporarily fix the issues.

Or maybe after replugging the monitor takes longer to display the screen so I miss the pop-up messages. I need to test this tomorrow.

juangentile, 2024-05-13:
@popcornmix in my case the problem is very clear when I log out. In the log in screen, if I switch the screen input (or switch input in kvm) then when going back to the pi the log in screen is all black and all I see is the cursor.

popcornmix, 2024-05-17:
Does the issue occur with [labwc](https://forums.raspberrypi.com/viewtopic.php?p=2220146#p2220146)?