raspberrypi/bookworm-feedback #148: Scratch 3 Raspberry Pi GPIO extension does not work on RPi5

penguintutor, 2023-11-18:
Running Scratch 3 on a Raspberry Pi 5; the GPIO extension does not work.

It is not correctly detecting any of the GPIO ports. 

I have tested using Bookworm on a Raspberry Pi 4 which works correctly, and I have also tested wiring of the GPIO on the RPi5 using the same ports with GPIOZero.

To recreate this then you can start with a basic example:

[image]

This is a basic example showing simple code to test which is added to the default sprite. GPIO 2 is connected via a switch to ground.

When run the Scratch cat will move across the screen whether the button is pressed or not. It should only move when the button is pressed.

I have also tested setting a GPIO pin high to drive an LED which does not work either.

aallan, 2023-11-24:
Duplicate of #91

penguintutor, 2023-11-24:
This is not related to the 16KB segmentation fault. 

I am able to get Scratch running both on the 32-bit Raspberry Pi OS. I've also got it working using the 64-bit version using kernel8.img. 

It is the GPIO extension which is not working and always gives a "high" when set to an input, and is not working when set as an output. 

I suspect this is that the GPIO extension is not aware of the changes to the GPIO using the RP1 and so may need to be updated

SivaVasudevan0204, 2023-11-28:
same for me...i can use other futures in Scratch3 with RPi5...but couldnt use these extension Raspberry Pi GPIO and Raspberry Pi Sense HAT. Anyone can help me out ?

grega, 2024-02-22:
The latest 3.30.8 release fixes issues loading the GPIO and Simple Electronic extensions:

64-bit: https://archive.raspberrypi.org/debian/pool/main/s/scratch3/scratch3_3.30.8_arm64.deb

32-bit: https://archive.raspberrypi.org/debian/pool/main/s/scratch3/scratch3_3.30.8_armhf.deb

SenseHAT issues have a fix incoming...