raspberrypi/bookworm-feedback #149: Screen sharing fails in Raspberry Pi OS

probonopd, 2023-11-18:
**Describe the bug**
Screen sharing fails in Raspberry Pi OS.

**To reproduce**
Flash Raspberry Pi OS and run it without any changes on Raspberry Pi 5.
Then try screen sharing using one of the two factory-provided web browsers, e.g., using https://meet.jit.si/.

**Expected behaviour**
Screen sharing works.

**Actual behaviour**

* In Firefox only sharing the full screen works, but you cannot share individual windows.
* In Chromium, you can only share Chromium tabs but neither individual windows nor the whole screen

Here is the full experience on a factory fresh Raspberry Pi OS image:
https://www.youtube.com/watch?v=6gDzZbImFJY

I suspect these issues are caused by Wayland, which is the default session on Raspberry Pi OS, and is known for breaking screen sharing.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Raspberry Pi 5 4GB RAM`
* Which OS and version (`cat /etc/rpi-issue`)? `Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4`
* Which firmware version (`vcgencmd version`)? `2023/10/18 18:30:17`
* Which kernel version (`uname -a`)? `Linux raspberrypi 6.1.0-rpi4-rpi-2712 raspberrypi/userland#1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux`

popcornmix, 2023-11-18:
I've transferred this issue to bookworm-feedback.

probonopd, 2023-11-18:
Thanks @popcornmix.

I had looked for a project named "raspberry-pi-os" or similar. Maybe renaming "bookworm-feedback" would help others, too?

sallyhaj, 2024-07-08:
Last time I tested Raspberry Pi OS with default GUI (Wayfire) on Raspberry Pi 5b, it seems sharing screen is okay.
But when it comes to use another desktop environment e.g. Gnome, that doesn't work, and almost have the same issue as @probonopd mentioned.
With sharing tab, it's okay:
[image]
With sharing window has problem:
[image]
With sharing full screen has problem as well:
[image]

sallyhaj, 2024-07-08:
For more information, when I tried sharing screen from chromium I got:
`[2:2:0707/210734.551088:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.ScreenSaver.GetActive: object_path= /org/freedesktop/ScreenSaver: org.freedesktop.DBus.Error.NotSupported: This method is not implemented
[2:15:0707/210735.038303:ERROR:ev_root_ca_metadata.cc(162)] Failed to decode OID: 0
[34:34:0707/210751.850181:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[34:34:0707/210751.851681:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
[34:34:0707/210757.085650:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
[2:170:0707/210803.727287:ERROR:egl_dmabuf.cc(632)] Failed to bind DMA buf framebuffer
[2:170:0707/210803.727339:ERROR:shared_screencast_stream.cc(938)] Dropping DMA-BUF modifier: 72057594037927935 and trying to renegotiate stream parameters
'loop->recurse > 0' failed at ../src/pipewire/thread-loop.c:426 pw_thread_loop_wait()
'loop->recurse > 0' failed at ../src/pipewire/thread-loop.c:426 pw_thread_loop_wait()`

And when I tried that from firefox, I got this:
`[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.

'loop->recurse > 0' failed at ../src/pipewire/thread-loop.c:425 pw_thread_loop_wait()
'loop->recurse > 0' failed at ../src/pipewire/thread-loop.c:425 pw_thread_loop_wait()`

cillian64, 2024-07-09:
> In Firefox only sharing the full screen works, but you cannot share individual windows.
> In Chromium, you can only share Chromium tabs but neither individual windows nor the whole screen

Both of these are expected with wayfire/labwc, as linked from here https://github.com/raspberrypi/bookworm-feedback/issues/149#issuecomment-1841954176, it's complicated.

> But when it comes to use another desktop environment e.g. Gnome, that doesn't work, and almost have the same issue as @probonopd mentioned.

We don't officially support gnome so I can't spend much time digging into this, but it looks like what's going wrong is the dmabuf buffers being shared by GNOME are in a GPU tiled format (which is more efficient for the GPU to read and write).  But webrtc doesn't know how to read these formats, so it gets all garbled.

This looks similar to https://askubuntu.com/questions/1483678/has-anyone-actually-managed-to-use-zoom-screen-share-on-wayland-strictly-no-x11 ,  so it might help to uninstall xdg-desktop-portal-wlr (which implements screen sharing on wlroots-based compositors like labwc and wayfire) and install xdg-desktop-portal-gnome (which implements screen sharing for gnome).  Otherwise, if you can convince gnome to use software rendering instead of hw-accelerated (i have no idea how) then that might fix it since that will use vanilla linear buffer formats.

sallyhaj, 2024-07-09:
I have asked about screen sharing in #wayland on IRC oftc, and they suggest the issue is related to mutter in gnome, and they asked me to try the new release of it, so I went ahead to Ubuntu24.4 which has mutter46, and it's indeed, screen sharing has no issue.
The problem with Debian Bookworm, as it has an old mutter43.8.
Thank you.

cillian64, 2024-07-10:
Thanks for the update, good to know it's not anything we've broken specifically.