raspberrypi/bookworm-feedback #152: Raspberry Pi GPIO and Raspberry Pi Sense HAT from Scratch3 doesn`t work on Raspberry Pi 5 32 bit and 4kb size pages

SivaVasudevan0204, 2023-11-24:
Hello, im trying to use Raspberry Pi GPIO and Raspberry Pi Sense HAT from Scratch3 in Raspberry Pi 5. but its didnt response. Btw Im using 32 bit and 4kb size pages. Can anyone help me out?

aallan, 2023-11-24:
Duplicate of #91

SivaVasudevan0204, 2023-11-27:
i think this is not related to #91. because i can use other features of scratch3 in RPi5...seems the extension Raspberry Pi GPIO and Raspberry Pi Sense HAT is only not working for me in RPi5. btw im using 32 bit and 4kb size pages. Can help me to solve this
[image]

SivaVasudevan0204, 2023-11-27:
[image]
im trying to display the A text..but the led just on like rainbow only..didnt even change since i plugin the sense Hat in my RPi5

lurch, 2023-11-27:
Duplicate of #148