raspberrypi/bookworm-feedback #156: PiOS Bookworm Window Issue

Technonyte0001, 2023-12-03:
I've recently downloaded Raspberry Pi OS Bookworm on my Pi 400, and everything works fine.

Until I open Chromium and try to un-check the "Use system title bar and borders" button. Doing this causes the os titlebar t hover slightly about the window, a bit displaces to the left, not hidden. Firefox'es works fine, but I'd like to use Chromium.

This same issue occurs when I create a Chromium shortcut and it opens as a new window. The same title bar weirdness occurs with the shortcut window.

This error might seem minimal compares to some others, but I would appreciate if this got patched.

Is it possible to fix this issue?

Thanks,
Technonyte

lurch, 2023-12-04:
This is what I'm seeing:
[image]

[image]

What's the problem?

Technonyte0001, 2023-12-05:
on the update before, like usual, it would remove the original os title bar, and add its own close, maximize and minimize buttons, for chromium, but in this update, (im not the only one who experiences this), it adds the chromium window controls, but the os title bar hovers slightly above the window, displaced to the left. it looks rather ugly. that is the problem.

ConstantMan, 2024-04-08:
I experience this too