raspberrypi/bookworm-feedback #166: Missing dependencies for VLC

djbuldog, 2023-12-10:
I cannot install VLC due to missing following dependencies:

````
#apt install vlc
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libvlccore9 : Depends: libidn11 (>= 1.13) but it is not installable
               Recommends: libproxy-tools but it is not going to be installed
 vlc-plugin-base : Depends: libflac8 (>= 1.3.0) but it is not installable
                   Depends: libprotobuf-lite23 (>= 3.12.4) but it is not installable
 vlc-plugin-video-output : Depends: libplacebo72 (>= 0.4.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
````
It looks that mentioned libraries were replaced by newer versions in the repository:
- libidn11 -> libidn12
- libflac8 -> libflac12
- libprotobuf-lite23 -> libprotobuf-lite32

My apt source is:
`deb http://ftp.fi.muni.cz/pub/linux/raspbian/raspbian bookworm main contrib non-free rpi`

Would be possible to rebuild VLC package to fix dependencies? or.. can I help with it / do it by myself somehow?
Sadly, I can't even install VLC from old stable due to other missing dependencies :(

cmitu, 2023-12-11:
Is this a new installation or an upgrade ? `libidn11` doesn't exist in any repository in a default RaspiOS _bookworm_ installation. It's also not present in the upstream (Debian) _bookworm_ repositories.
Can you run `apt-cache policy vlc` and show the result ?

lurch, 2023-12-11:
I don't know what setup you have, but the Desktop images of Raspberry Pi OS available from https://www.raspberrypi.com/software/operating-systems/ all come with VLC pre-installed, so there's obviously no dependency problems there. As a test, I even tried running this sequence of commands:
```
sudo apt purge vlc
sudo apt autoremove
sudo apt install vlc
```
and it all worked fine, so you must be doing something very non-standard. Closing as `wontfix`, but will re-open if you can demonstrate an actual problem with Raspberry Pi OS Bookworm.

djbuldog, 2023-12-11:
You are right. The problem was in my installation. I increased version to bookworm in /etc/apt/sources.list, but there was another reference to previous bullseye in /etc/apt/sources.list.d/raspi.list. I didn't notice the other config file, sorry.

````
# apt-cache policy vlc
vlc:
  Installed: (none)
  Candidate: 1:3.0.20-0+rpt1+deb11u1
  Version table:
     1:3.0.20-0+rpt1+deb11u1 500
        500 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
     3.0.20-0+deb12u1+rpi1 500
        500 http://ftp.fi.muni.cz/pub/linux/raspbian/raspbian bookworm/main armhf Packages
````