raspberrypi/bookworm-feedback #168: Total system freeze instead of malloc call failure

qrp73, 2023-12-12:
Steps to reproduce:
1) create file main.cpp:
```
#include <stdio.h>
#include <errno.h>
#include <string.h>
#include <cstdlib>
#include <stdint.h>

// g++ -std=c++14 main.cpp -o testmem

int main() {
    const auto allocSize = 1024 * 1024; // 1 MB block
    const auto allocCount = 32 * 1024;  // 32768 blocks

    uint8_t* memv[allocCount];
    memset(memv, 0, sizeof(memv));

    for (auto i=0; i<sizeof(memv)/sizeof(uint8_t*); i++) {
        printf("malloc %04d MB..", i+1);
        uint8_t* pbuf = (uint8_t*)malloc(allocSize);
        if (pbuf == NULL) { 
            printf("failed\n");
            fprintf(stderr, "malloc failed, errno: %d\n", errno);
            break;
        }
        printf("OK, memset..");
        //memset(pbuf, 0, allocSize);
        for (auto j=0; j < allocSize; j++) {
            pbuf[j] = 0;
        }
        printf("OK\n");
        memv[i] = pbuf;
    }
    printf("Press any key to release memory and exit...\n");
    getchar();

    for (auto& pbuf : memv) {
        if (pbuf != NULL)
            free(pbuf);
    }

    return 0;
}
```
2) compile it with:
```
$ g++ -std=c++14 main.cpp -o testmem
```
3) run it from terminal in wayland window:
```
$ ./testmem
```

Expected result: when process allocates all available memory, malloc returns NULL to indicate error and app shows error code. The system continue to work and user is able to kill the process.

Actual result: the system completely freezes, mouse stops to move, keyboard don't responds. With no way to kill process and the only way to exit is hardware power off for reset.

Additional findings: 
1) when process is started from virtual terminal (Ctrl+Alt+F1) it also leads to system freeze but allows to stop process with Ctrl+C
2) when this code:
```
        for (auto j=0; j < allocSize; j++) {
            pbuf[j] = 0;
        }
```
is replaced with this one:
```
memset(pbuf, 0, allocSize);
```
it leads to process auto kill when out of memory is reached. 
I'm not sure why user code which fills memory with zero don't works the same as calling to memset?

The issue was caught with many apps which requesting too many memory, when running in wayland desktop mode it leads to a total system freeze with no way to stop it.

qrp73, 2023-12-12:
Just found workaround. Pressing Alt+SysRq+F kills process which eat all memory and system unfreezes.

pelwell, 2023-12-12:
What do you get after disabling swap (`sudo swapoff -a`)?

popcornmix, 2023-12-12:
I think this is linux's default overcommit behaviour. See [here](https://stackoverflow.com/questions/16674370/why-does-malloc-or-new-never-return-null)

Try running this first:
```
echo 2 | sudo tee /proc/sys/vm/overcommit_memory
```

qrp73, 2023-12-12:
> What do you get after disabling swap (`sudo swapoff -a`)?

The swap is disabled on my system. 
With disabled swap it leads to a total system freeze, when there is not enough memory on the system, the malloc call never returns and waits for infinite.
The only way to resume it is to use Alt+SysRq+F to kill process.

pelwell, 2023-12-12:
It looks like @popcornmix's suggestion might be a better one.

qrp73, 2023-12-12:
> What do you get after disabling swap (`sudo swapoff -a`)?

The swap is disabled on my system. 
With disabled swap it leads to a total system freeze, when there is not enough memory on the system, the malloc call never returns and waits for infinite.
The only way to resume it is to use Alt+SysRq+F to kill process.

> I think this is linux's default overcommit behaviour. See [here](https://stackoverflow.com/questions/16674370/why-does-malloc-or-new-never-return-null)
> 
> Try running this first:
> 
> ```
> echo 2 | sudo tee /proc/sys/vm/overcommit_memory
> ```

just tried it, but after this command there is no way to execute any command anymore and after 10-15 seconds wayland unexpectedly stopped and login screen appears, but attempt to login fails due to unexpectedly stopped process.

pelwell, 2023-12-12:
What are you running your test on? Some platforms are already quite memory-constrained, e.g. Zero 2 W (especially running a 64-bit kernel).

qrp73, 2023-12-12:
I run it on RPI 4B rev  1.5 4GB with Bookworm and latest kernel rpi-6.6.y.
```
$ uname -a
Linux raspi 6.6.6-v8+ #1 SMP PREEMPT Mon Dec 11 15:52:35 UTC 2023 aarch64 GNU/Linux
```

Just tried ```echo 2 | sudo tee /proc/sys/vm/overcommit_memory``` again after reboot with enabled swap, now after executing this command firefox unexpectedly stopped, attempt to start it again fails:
```
$ free
               total        used        free      shared  buff/cache   available
Mem:         3883128      473856     2295916       46284     1237092     3409272
Swap:              0           0           0
$ firefox
[GFX1-]: glxtest: DRM render node not clearly detectable. Falling back to using the only one that was found.
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.

out of memory: 0x0000000000000012 bytes requested
<jemalloc>: (malloc) Error initializing arena
```

attempt to run the test code from a first message in this issue leads to fail and SIGKILL after 290-490 MB allocation, despite the fact that 2-3 GB memory is free.

ghollingworth, 2023-12-22:
I think originally the code just does standard normal Linux practice, which is to say it tries its hardest to allocate the memory you've asked for.  But as you get to the limits of what it can do it then has to manipulate a lot of other things (caches, de-fragmentations etc) to release more and more memory.  It keeps doing this (while the computer becomes unresponsive) for quite a while.  You can see this if you run your code from an ssh terminal.

If you stop Linux from over committing its memory, it'll fail...  But Linux won't just let a single process allocate every byte of memory...  It'll stop when the user hits /proc/sys/vm/user_reserve_kbytes which seems to be 131072 (i.e. 1.3Gbytes)

If you:

echo 1 | sudo tee /proc/sys/vm/oom_kill_allocating_task

You'll find it does something more like what you were expecting.

But in the meantime you will always get a system freeze while the kernel does everything it can to try to reallocate memory!

closed 2023-12-22 by ghollingworth

qrp73, 2023-12-22:
yes, original behavior looks like standard linux behavior, the only issue here is that wayfire/pcmanfm is immediately freezed when out of memory happens. I think it is possible to keep it running by optimize memory allocations (avoid continuous requests for memory allocation).

But with ```echo 2 | sudo tee /proc/sys/vm/overcommit_memory``` it should works ok, isn't it? But it leads to all processes crash and unable to allocate memory despite the fact that it is available... I think this is not normal behavior, isn't it?