raspberrypi/bookworm-feedback #169: Excessive GPU-usage when wiggling the mouse in VSCode in 64-bit Bookworm

lurch, 2023-12-12:
I found a weird problem with VS Code being unexpectedly slow / laggy on a Raspberry Pi 5 running 64-bit Bookworm, and I've managed to narrow it down to a "minimal test case".

The "working" setup:
1. Flash `2023-12-05-raspios-bookworm-armhf.img.xz` (from https://www.raspberrypi.com/software/operating-systems/ ) to a micro-SD card
2. Boot it up in a Pi5, and apply all APT-updates
3. Install "Visual Studio Code" from the "Recommended Software" application
4. Enable the CPU usage and GPU usage taskbar plugins
5. Reboot, just for good measure
6. Open up VS Code
7. Maximise the window
8. Wait for the CPU and GPU usage monitors in the taskbar to flatten down to idle
9. Rapidly jiggle the mouse up and down over the 5 icons in the top-left of the window
10. The CPU usage hovers around 15% usage, GPU usage hovers around 0% usage

The "problematic" setup:
If I repeat all of the above using `2023-12-05-raspios-bookworm-arm64.img.xz` (i.e. the 64-bit image instead of the 32-bit image), then in the final step the CPU usage hovers around 15% usage, but the GPU usage shoots up to 80% !!!

@6by9 looked at this with me, and setting `kernel=kernel8.img` in the 64-bit Bookworm's `config.txt` seems to have no impact on the problem (i.e. it's still bad).

I've just run the above usage-scenario on a Pi 4 too, and that also shows excess GPU-usage on 64-bit Bookworm but minimal GPU-usage on 32-bit Bookworm.

qrp73, 2023-12-13:
I'm using vscodium with clangd, CMake Tools and CodeLLDB plugins and there is no excessive GPU usage. 
It has some minor CPU usage, but this is expected behavior because it needs to scan and parse code in background...

lurch, 2023-12-13:
After digging into this a bit more with help from some of the other engineers here, it looks like this was a red herring! :facepalm: 
The problem actually seems to be that the GPU usage monitor is reporting incorrect results on 32-bit Bookworm (which I've raised as an internal issue).

closed 2023-12-13 by lurch