raspberrypi/bookworm-feedback #170: Regression: Broken button icons on gnome apps and gedit

qrp73, 2023-12-13:
It happens about a week ago after some apt update, button icons disappeared and now there is just empty space. 
I thought it will be noticed and fixed soon, but it seems that no one notice it yet...

Yesterday I also tested it on a clean Bookworm aarch64 install with latest updates and this bug is reproduced.

## Steps to reproduce:
1) install gedit and some gnome apps:
```
sudo apt update && sudo apt install gnome-calculator gnome-clocks gedit
```
2) Open installed apps

Expected result: button icons is properly displayed 

Actual result: button icons are missing, empty space is displayed (it worked ok about 2-3 weeks ago before some update)

## Screenshot
[image]

lurch, 2023-12-13:
It looks like this also affects the installed-by-default mousepad application.

1. Launch Accessories -> Text Editor
2. Click File -> New
3. Missing close-icons on the tabs:
[image]
[image]
[image]
(the tooltips appear when I mouse-hover over where the close-icons _would_ be)

spl237, 2023-12-13:
OK, fix now uploaded to apt - should be available once things have rebuilt in the repos.

And now I am definitely on Christmas holiday - I only fixed this one now because I was a bit bored... ;)

spl237, 2023-12-14:
Changes to fix this are now in apt - updating should fix it.

closed 2023-12-14 by spl237

lurch, 2023-12-14:
Thanks Simon. That fixes all the "close" icons, but there's still two icons (I guess these might be custom minimise & maximise icons) that are missing for gnome-clocks and gnome-calculator.
[image]

Enjoy your holiday! :christmas_tree:

spl237, 2023-12-14:
Oh for crying out loud - bloody Gnome! How the hell have they managed to do that? That's presumably a completely custom header bar which is neither the GTK one, the libhandy one, nor one created using standard buttons. None of those were good enough, in spite of the fact that they designed all of them - no, let's create yet another one. Give me strength...

I'm half-inclined to leave that broken, frankly - they aren't even obeying their own standards. I suspect a similar hack will fix it, though - I'll have a look.

spl237, 2023-12-14:
OK, newly patched versions of relevant packages now uploaded to the repo - should appear in a few hours. I've made the minimize and maximize icons behave exactly the same as the close icon, so they now have separate overrides if needed.

lurch, 2023-12-18:
Can confirm that this is all displaying correctly now :+1: