raspberrypi/bookworm-feedback #171: chromium-browser: jerking video playback

qrp73, 2023-12-13:
Some videos are played with jerking. It looks like some fragments are repeatedly played several times in a loop and then jump to next fragment. It looks like some issue with video decoder.

For example: https://t.me/wildlifen/2263

ghollingworth, 2023-12-15:
I don't get this, for me it works fine...

Can you reproduce with a clean image?

qrp73, 2023-12-15:
Just tried it on clean Bookworm install (with latest apt update && apt upgrade).
Yes, I can stable reproduce it on clean Bookworm install.

The issue is reproduced with default arm/gpu clock (no overclocking) and is reproduced with overclocked arm=2100/gpu=700. Also tested with different display refresh rate settings:  60 Hz and 75 Hz, it is reproduced on both, but at 60 Hz it a little bit less flicker.

## Hardware info
**Device:** Raspberry Pi 4B 4G rev 1.5 + genuine Raspberry Pi 15W PSU

```
$ vcgencmd version
Dec  8 2023 12:36:15 
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start)

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ uname -a
Linux raspi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

qrp73, 2023-12-15:
Also tried to open this video on firefox with enabled hardware acceleration, but it appears that it still playing it at low fps. 

But there is interesting finding. When hardware acceleration is enabled, playing this video on firefox leads to a very high lags. Mouse cursor is very lagging. Monitor shows load GPU=96% and CPU=18%.

With disabled hardware acceleration it shows the same fps, but no mouse lags and monitor shows GPU=3% and CPU=40%.

So, something with hardware acceleration is definitely going wrong.

I tested it on this page: https://t.me/wildlifen/2263

qrp73, 2023-12-15:
Additional and important finding!

When I open page https://t.me/wildlifen/2263 in chromium browser and hover mouse cursor over New Tab button (on the right side of the tab header) flickering and jerky movement disappears and video is played very smooth (as it should). 

But once I move mouse cursor outside New Tab button area, it starts to flicker and jerky playback again.

qrp73, 2023-12-15:
Just tried to write bookworm image on a new sdcard and tested it on a clean install with all default settings. 
The issue is present.

lurch, 2023-12-18:
I've been able to reproduce this... (all tests below are on a fully up-to-date 64-bit Bookworm image)

On a Pi5 in Chromium, the video plays buttery-smooth (so I suspect this is what Gordon was seeing) and it uses 20% CPU and 84% GPU.
On a Pi5 in Firefox, the video also plays smoothly, and uses 12% CPU and 95% GPU (yikes?!)

On a Pi4 in Chromium I get the jerky playback (it kinda looks like the frame-count is skipping both backwards and forwards) and it uses 31% CPU and 85% CPU.
On a Pi4 in Firefox I get a low framerate, but I don't get the skipping-backwards-and-forwards behaviour; it uses 15% CPU and 97% CPU

On a Pi3B+ in Chromium the playback is such a low frame-rate, it looks like https://www.raspberrypi.com/news/watch-a-festive-film-every-second-of-the-day-until-christmas/
On a Pi3B+ in Firefox it's still a low frame-rate, but much better than Chromium on the same hardware.

ghollingworth, 2023-12-19:
It looks like there is javascript that is forcing the re-drawing of the whole page (not just the video) for every frame...

So I think this is mostly just a case of "bad webpage"

lurch, 2023-12-19:
We've chatted about this internally, and decided to close it as "wontfix" as it's simply a badly-written webpage.

closed 2023-12-19 by lurch

qrp73, 2023-12-19:
But why this page is played ok on PC? 
Why it is played ok when a hint window appears? 

It seems like video driver issue, because it don't depends on js behavior, but depends on OS GUI state (window hint). Isn't it?

lurch, 2023-12-19:
> But why this page is played ok on PC?

One of the other engineers here told me "On my Windows home PC, in Firefox this uses ~25% of my rather beefy desktop GPU."

> Why it is played ok when a hint window appears?

I dunno, that's probably a question for the Chromium developers...

ghollingworth, 2023-12-19:
David said:

> I've had a quick poke around and I think what's happening (it's hard to tell with the obfuscated JS) is the page tries to redraw the page background on scroll events, but in doing so it triggers scroll events, and so most of the page ends up getting redrawn constantly.  It goes away when the page doesn't have focus, I assume because that disables scroll events.  Also, even if you delete the video in the DOM inspector, the rest of the (static) page still uses masses of GPU.

So when the tooltip appears I'm guessing those scroll events disappear as well!