raspberrypi/bookworm-feedback #172: Strange button-colours in LibreOffice in 64-bit Bookworm

lurch, 2023-12-13:
With freshly-downloaded "Full" OS images from https://www.raspberrypi.com/software/operating-systems/

In 32-bit Bookworm, the toolbar icons in LibreOffice Calc and Writer look correct (black lines on a light-grey background) :
[image]

but in 64-bit Bookworm, the toolbar icons are drawn with white lines on a light-grey background, which looks very odd :
[image]

pelwell, 2023-12-13:
What happens when you activate the new Dark theme?

lurch, 2023-12-13:
Haha, with the Dark theme the situation is reversed, with the 64-bit image then looking "good" but the 32-bit image looking "bad".

32-bit image shows black lines on dark-grey background :
[image]

64-bit image shows whie lines on dark-grey background :
[image]

spl237, 2023-12-13:
OK, so for some reason, the 64-bit image is showing a different set of icons from the 32-bit image.

I have absolutely no idea how that can even happen, unless the 64-bit image has different build flags.

Given I don't think we change the libreoffice package ourselves any more, I suspect this is one for upstream, but I will check in the New Year.

spl237, 2023-12-14:
OK, for some reason different builds of LibreOffice are loading different default icon themes. We just pull LibreOffice straight from Debian - we don't customise it ourselves - so this is an upstream issue and not anything we have broken, as far as I can tell.

You can manually select the icon theme you want - Tools menu, Options, View, then choose from the Icon Theme dropdown. Looks like you will need to change that depending on whether you are running 32 or 64 bit!

lurch, 2023-12-14:
Yeah, strange. On 32-bit Bookworm it says:
```
$ apt policy libreoffice-writer
libreoffice-writer:
  Installed: 4:7.4.7-1+rpi1
  Candidate: 4:7.4.7-1+rpi1
  Version table:
 *** 4:7.4.7-1+rpi1 500
        500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
```
and defaults to the "Automatic (Colibre)" theme.

Whereas on 64-bit Bookworm it says:
```
$ apt policy libreoffice-writer
libreoffice-writer:
  Installed: 4:7.4.7-1+deb12u1
  Candidate: 4:7.4.7-1+deb12u1
  Version table:
 *** 4:7.4.7-1+deb12u1 500
        500 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages
        100 /var/lib/dpkg/status
     4:7.4.7-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
```
and defaults to the "Automatic (Colibre (SVG + dark))" theme.

And just to confirm, if I set LibreOffice on 64-bit Bookworm to the "Colibre" theme, then the toolbar-buttons look "correct" (i.e. the same as they do on 32-bit Bookworm). And the good news is that if you change the theme in LibreOffice Writer, it also changes the theme in LibreOffice Calc :slightly_smiling_face:

spl237, 2023-12-14:
Ah - so the Raspbian one is being patched by plugwash - hence the +rpi1. Maybe that is the cause of the change, but it's not something I can do anything about!