raspberrypi/bookworm-feedback #173: Mouse input responsiveness issues and and mouse cursor visual glitches on RPI-5

sudo-splinter-cell, 2023-12-14:
Platform: RPI-5 (4GB)
OS: RPIOS (bookworm)
Desktop: KDE Plasma (but the experience is similar in other desktop environments)

When X11 is selected as the display backend, mouse cursor constantly flickers and visually glitches and even becomes invisible unless you move the mouse. This can be resolved by disabling the screen compositor (if possible)

When Wayland is selected as the display backend, the responsiveness of mouse cursor to the mouse movements is delayed, especially if the Pi is under light to moderate load. Under heavy load, it is really a struggle to controll the mouse cursor. And the visual reflection of the mouse cursor on the screen especially to the mouse movements is in "lower fps" than the screen refresh rate. So when you move the mouse fast, it feels like a low fps reaction and the mouse "skips" some parts of the screen. On X11, no matter how fast you move the mouse, the cursor is always clearly visible.

lurch, 2023-12-14:
Similar to #65 ?

popcornmix, 2023-12-14:
I think this post has included two unrelated issues. One of which is a duplicate of #65.

The visual glitches is more interesting. [This post](https://forums.raspberrypi.com/viewtopic.php?p=2168692#p2168692) referred to "occasionally a big black rectangle appears".

I think that is worth investigating, but we need instructions for how to reproduce.

ghollingworth, 2023-12-15:
Sounds like this is a similar problem to elsewhere, where the issue is KDE is using the GPU so heavily it is not available for composition (you can only use it in one place at a time!)...

It would be interesting to see if it works better if you stop KDE from using the GPU, but not sure how to do that...

Trotter73, 2023-12-15:
Hi,  I'm the bod who has been having issues with Cinnamon..

Firstly to replicate, I have been starting with a Lite Bookworm image, update & upgrade and then using tasksel to install the Cinnamon desktop. You can also start from the desktop image, the end result, flickering, is the same but you have to change the greeter by messing about with the lightdm.conf file

sudo tasksel

Select the Cinnamon desktop and wait.

When Cinnamon has installed you also need to install gldriver-test , when starting from the lite image you cannot or at least couldnt boot to the GUI without it, the boot process would appear to stall, there are threads about this....

sudo apt install gldriver-test 

Set a GUI boot

sudo systemctl set-default graphical.target

Now reboot and you SHOULD comeup with the Cinnamon desktop, it's been a couple of weeks since I tried this, updates may have changed the situation....

When in the DE open a few windows, a shell is a good one and you will notice the mouse pointer will regularly flash, as described above, of you hover around top right or lower right of a window the pointer may well start to flicker a lot quicker or even disappear, it may take a few moments for this to happen, you add some desklets something like XKCD or a weather app, you may find the flashing gets worse...

The black rectangle has been described above by @sudo-splinter-cell I have found the same but not a specific trigger for that, anecdotally maybe when the CPU is a bit busy...

I also noticed moving windows about uses more CPU than I would have imagined if everything as on the GPU, hearsay  on this though I'm afraid dont have anything to back it up....

@ghollingworth With reference to KDE, when running on Wayland it seems very good none of these graphical issues,  at the login screen (SDDM) you get some strange pointer effects 3 pointers and some corruption but generally better. I've not tried Plasma under X11, I can have a go if you think it will help, not sure I saw on option though if I'm honest.

popcornmix, 2023-12-15:
Is the black rectangle always transitory, or does it ever persist on screen once mouse movement has stopped?
A photo, or video (using a phone camera) that shows the black rectangle would be useful.

I'm curious if the black rectangle is 64x64 matching the mouse pointer overlay.
e.g.
```
pi@pi4:~ $ kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@30.00 297.000 3840/176/88/296/+ 2160/8/10/72/+ 30 (30.00) D 16:9
      Plane 3 (84) fb-id: 342 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 342 3840x2160 XR24
      Plane 25 (307) fb-id: 345 (crtcs: 3) 0,0 64x64 -> 1396,769 64x64 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 345 64x64 AR24
```
You can see here there is 64x64 AR24 format overlay at position 1396,769 on screen.

Trotter73, 2023-12-16:
Hi,

The black square/rectangle is transitory, found a way to reliably trigger it and the flickering cursor, vids should be shown below. The rectangle flashed quite quickly so I shot it in slomo....

As a side note I couldn't get Cinnamon to boot using the lite image as a base, something has changed, for the test below I used the desktop image as a base and installed Cinnamon over the top of that....

Hope this helps.

https://github.com/raspberrypi/bookworm-feedback/assets/16307645/a876728b-6c40-40a1-b47c-16d513b20bea

https://github.com/raspberrypi/bookworm-feedback/assets/16307645/5422c0ca-db2d-4e6e-a1c7-1109a11b0492

sudo-splinter-cell, 2023-12-18:
KDE Desktop with compositor enabled, this is the mouse cursor acting up. It loves to mimic the keyboard cursor :) If i turn off the compositor, mouse issues "mostly" go away.

https://github.com/raspberrypi/bookworm-feedback/assets/79761284/9b8745a9-bafc-4ad6-b0fa-210ea6355e30

The mouse cursor issue with wayland is different than this by the way.

sudo-splinter-cell, 2024-01-30:
I have switched to Wayland to see if anything was improved over time and pretty much all the issues still persist.